rzeldent/esp32-smartdisplay #2: Odd tearing on display

edasque, 2023-01-31:
Using the sample code at https://github.com/x821938/ESP32-2432S028R-arduino-platformio-simple-test that uses your drivers, I see odd artifacts on a ESP32-2432S028R. Any idea what could be causing it?

https://user-images.githubusercontent.com/219187/215879092-91acd49a-5d00-4e71-928f-2aeb31239679.MOV

rzeldent, 2023-01-31:
Hi Erik,

I had the same and in my case it was out of the bounds of the display.
LVGL adds some margins and if using the exact size of the display this
might happen.

I use:
screen = lv_obj_create(nullptr);
lv_obj_remove_style_all(screen);
on a screen so it will fit...

Hope this helps,

Rene

rzeldent, 2023-02-05:
Hi Erik,

Did it help, lowering the SPI freq?

edasque, 2023-02-06:
Sorry I was away for a few days. Neither worked, unfortunately. The frequency change nor changing the power supply (from my computer or a big power supply brick).

I'll try to run your code sample see if I can get that working well.

rzeldent, 2023-02-14:
Still strange, any updates/findings?

maxbrito500, 2023-03-10:
Hi Rene, same issue with the same model mentioned above.

Just to let you know, I will continue debugging to try solve this.

Previously the LovyanGFX sample code managed to write well on the display,

Also had the issues (solved):
+ montserrat font missing
+ knowing where to place lvgl_conf.h
+ getting started with the example

Your work is the most advanced on this model.  Perhaps it would be possible to release a version for the Arduino IDE in ino format?

Many thanks for making this available!

rzeldent, 2023-03-11:
Hi Max,

Still strange why this happens on some boards.. Maybe timing or another display?
Someone had also a board that had the RGB switched to BGR. Or maybe timing?
Please let me know if you find something. Maybe compare with LovyanGfx?
Will also take a look again if I have time...

Kind regards,

Rene

maxbrito500, 2023-03-11:
Hello Rene,

Can confirm the switched BGR leds on the more recent model (ordered February 2023), had the same situation. Last night was possible to achieve a fully working project with touchscreen, display color, LovyanGFX and LVGL. The code was uploaded to this location: https://github.com/radio3-network/kit-ESP32-2432S028R/tree/main/ESP32_2432S028-LovyanGFX-LVGL

In the meanwhile I've learned about the smartdisplay being a library on platformIO. I'm a newbie, didn't knew it was a library and it is difficult for me to start a new project there (also new to C language, albeit long time Java). Maybe possible to provide a working template project with an LVGL button demo?

Thanks for looking into this topic Rene. From the projects I've seen related to this board your code is by far the most advanced and complete. The board is so cheap and useful, just too bad about the (inexistent) library support. Maybe we can convince the manufacturer to point into your repository as location for others to start too.

rzeldent, 2023-03-12:
Hi Brito,

I made a demo project at https://github.com/rzeldent/esp32-smartdisplay-demo.
Think this will help you a lot!
Do not forget to define the settings in platformio.h!

Kind regards,

Rene

maxbrito500, 2023-03-12:
Fantastic Rene!

It worked like a charm. First time tried with RGB and had mixed colors, second time went with BGR as option on platformio.ini and this worked as expected. After that went to try functions like smartdisplay_get_light_intensity() and again worked as expected.

Man, felt great to just program code that runs on the board rather than hunting down error messages for hours.

Thank you for that Rene, made a big difference.

rzeldent, 2023-08-28:
You're welcome!

closed 2023-08-28 by rzeldent