rzeldent/esp32-smartdisplay #3: Blue and red inverted

edo44, 2023-02-09:
I have the two colors inverted for some reason when using LVGL. Any help?

rzeldent, 2023-02-09:
Hi Edo,

You can try to change the MADCTL value. This is currently defined like this (depending on the orientation)
static const uint8_t madctl[] = {MADCTL_MY | MADCTL_RGB}; // Portrait 0 Degrees

so change the color channels to BGR:
static const uint8_t madctl[] = {MADCTL_MY | **MADCTL_BGR**}; // Portrait 0 Degrees

Let me know if this helps. What are you using as display board?

edo44, 2023-02-09:
> 

yep, that solves the problem, thank you very much

ESP32_3248S035C

rzeldent, 2023-02-10:
Strange, I have also a ESP32_3248S035C and this worked with the MADCTL_RGB settings.
Maybe there are different panels? Maybe need to be configurable in the defines...

edo44, 2023-02-10:
> Strange, I have also a ESP32_3248S035C and this worked with the MADCTL_RGB settings. Maybe there are different panels? Maybe need to be configurable in the defines...

[image]

rzeldent, 2023-02-10:
[image]

rzeldent, 2023-02-10:
This is mine; they look identical. I suspect the LCD panel might be another type....
Maybe some other users can share if they've also this problem?

edo44, 2023-02-10:
[image]

rzeldent, 2023-02-10:
This is the capacitive touch controller

edo44, 2023-02-10:
> This is the capacitive touch controller

I see, I thought it was the tft screen controller

rzeldent, 2023-02-11:
Hi edo,

Made this setting an optional define and pushed a new version (1.0.4) to PlatformIO.

Enjoy!

closed 2023-02-11 by rzeldent