rzeldent/esp32-smartdisplay #6: Compile error

martin072, 2023-02-13:
Hi,

I am new to the display, was trying to compile your library, but it fails with the following error. Any idea?

```
.pio/build/esp32dev/src/main.cpp.o:(.literal._Z10mainscreenv+0x8): undefined reference to `lv_font_montserrat_22'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```

rzeldent, 2023-02-13:
Hi Martin,

You have to include the font in lv_conf.h.

Change the line
#define LV_FONT_MONTSERRAT_22 0
to
#define LV_FONT_MONTSERRAT_22 1

Regards

martin072, 2023-02-13:
Yup, that was  already done.. still an error:

```
Building in release mode
Compiling .pio/build/esp32dev/src/main.cpp.o
Linking .pio/build/esp32dev/firmware.elf
/Users/martin/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/src/main.cpp.o:
(.literal._Z10mainscreenv+0x8): undefined reference to `lv_font_montserrat_22'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```

rzeldent, 2023-02-13:
Hi Martin,

What lv_conf are you updating? Have you defined the include path in platformio.ini?

    # LVGL settings
    -D LV_CONF_PATH="${platformio.include_dir}/lv_conf.h"

martin072, 2023-02-13:
Hi Yes,  

Well as follows:

 -D LV_CONF_PATH="lv_conf.h"

its in the same dir. as main.cpp also there is no error about the lv_conf.h..

rzeldent, 2023-02-13:
I suggest to make the path absolute

martin072, 2023-02-13:
Thanks, that did the trick.

closed 2023-02-13 by martin072

iqbalelectro, 2024-07-17:
still dont understand, how to solve it? i already copy the lv_conf.h to the src folder where main.cpp is.

change f-D LV_CONF_PATH="${platformio.include_dir}/lv_conf.h" line at platformio.ini to 
-D LV_CONF_PATH="lv_conf.h"

am i still make a mistake?