rzeldent/esp32-smartdisplay #7: Display update issue

mihai-manea, 2023-02-14:
Hello,

After 2 days with a lot of errors, I finally manage to compile and upload your example.
But, I have an issue, the display is updating very slow (5-10sec). What could be the issue?

https://youtu.be/dcS2ea-2OOQ

rzeldent, 2023-02-14:
Hi Mihai,

I saw the video, very strange and also strange that the CPU remains at 0%. So, by watching it I cannot determine what's going on. I also did not hear from others about such an issue.
Could you share the code (with me) so I can test it on my hardware?

Rene

rzeldent, 2023-02-14:
Hi Mihai,

I was able to reproduce the issue. The delay is not caused by the library but by the 
```
 getLocalTime(&timeinfo);
```
I think when the time is not synchronized because there is no network or NTP it blocks.
I will update the sample to use millis().

Thanks for noticing!

closed 2023-02-14 by rzeldent