rzeldent/esp32-smartdisplay #9: undefined reference to `lv_font_montserrat_22'

petr-vytlacil, 2023-03-06:
Hi a try example demo from repository and get error when build:

.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o:(.literal._Z10mainscreenv+0x8): undefined reference to `lv_font_montserrat_22'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
==================================== [FAILED] Took 19.07 seconds ===

platformio.ini:

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

monitor_speed = 115200

build_flags =
    -Ofast
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    # LVGL settings
    -D LV_CONF_PATH="lv_conf.h"
    # Smartdisplay settings
    -D TFT_PANEL_ORDER_RGB
    #-D TFT_PANEL_ORDER_BGR
    #-D TFT_ORIENTATION_PORTRAIT
    -D TFT_ORIENTATION_LANDSCAPE
    #-D TFT_ORIENTATION_PORTRAIT_INV
    #-D TFT_ORIENTATION_LANDSCAPE_INV
    -D ESP32_2432S028R
    #-D ESP32_3248S035R
    #-D ESP32_3248S035C

lib_deps =
    rzeldent/esp32_smartdisplay@^1.0.4

thx

rzeldent, 2023-03-06:
Hi Petr,

In the lv_conf.h, that is included the following define must be present to
include the (missing) font.
#define LV_FONT_MONTSERRAT_22 1

This error is often created because the path to the lv_conf is not absolute.
See issue #6: https://github.com/rzeldent/esp32-smartdisplay/issues/6

Kind regards,

Rene

petr-vytlacil, 2023-03-06:
I'm sorry, I don't understand.
In project i use lv_conf.h from your example. 

There is log from build. 
THX

`.c:10:
.pio/libdeps/esp32dev/lvgl/src/extra/widgets/colorwheel/../../../../src/misc/../lv_conf_internal.h:46:17: note: #pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors
         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
                 ^~~~~~~
In file included from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/keyboard/../../../widgets/lv_btnmatrix.h:16,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/keyboard/lv_keyboard.h:16,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/keyboard/lv_keyboard.c:10:
.pio/libdeps/esp32dev/lvgl/src/extra/widgets/keyboard/../../../widgets/../lv_conf_internal.h:46:17: note: #pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors
         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
                 ^~~~~~~
In file included from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/led/../../../../src/misc/lv_log.h:16,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/led/../../../../lvgl.h:25,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/led/../../../lvgl.h:17, 
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/led/lv_led.h:16,        
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/led/lv_led.c:9:
.pio/libdeps/esp32dev/lvgl/src/extra/widgets/led/../../../../src/misc/../lv_conf_internal.h:46:17: note: #pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors
         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
Compiling .pio\build\esp32dev\lib87c\lvgl\extra\widgets\menu\lv_menu.c.o
                 ^~~~~~~
In file included from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/imgbtn/../../../../src/misc/lv_log.h:16,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/imgbtn/../../../../lvgl.h:25,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/imgbtn/../../../lvgl.h:17,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/imgbtn/lv_imgbtn.h:16,  
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/imgbtn/lv_imgbtn.c:10:  
.pio/libdeps/esp32dev/lvgl/src/extra/widgets/imgbtn/../../../../src/misc/../lv_conf_internal.h:46:17: note: #pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors
         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
                 ^~~~~~~
Compiling .pio\build\esp32dev\lib87c\lvgl\extra\widgets\meter\lv_meter.c.o
Compiling .pio\build\esp32dev\lib87c\lvgl\extra\widgets\msgbox\lv_msgbox.c.o
Compiling .pio\build\esp32dev\lib87c\lvgl\extra\widgets\span\lv_span.c.o
Compiling .pio\build\esp32dev\lib87c\lvgl\extra\widgets\spinbox\lv_spinbox.c.o
In file included from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/list/../../../core/lv_obj.h:16,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/list/lv_list.h:16,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/list/lv_list.c:9:       
.pio/libdeps/esp32dev/lvgl/src/extra/widgets/list/../../../core/../lv_conf_internal.h:46:17: note: #pragma message: Possible failure to include lv_conf.h, please read the comment in 
this file if you get errors
         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
                 ^~~~~~~
Compiling .pio\build\esp32dev\lib87c\lvgl\extra\widgets\spinner\lv_spinner.c.o
Compiling .pio\build\esp32dev\lib87c\lvgl\extra\widgets\tabview\lv_tabview.c.o
Compiling .pio\build\esp32dev\lib87c\lvgl\extra\widgets\tileview\lv_tileview.c.o
In file included from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/menu/../../../core/lv_obj.h:16,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/menu/lv_menu.h:16,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/menu/lv_menu.c:9:       
.pio/libdeps/esp32dev/lvgl/src/extra/widgets/menu/../../../core/../lv_conf_internal.h:46:17: note: #pragma message: Possible failure to include lv_conf.h, please read the comment in 
this file if you get errors
         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
                 ^~~~~~~
Compiling .pio\build\esp32dev\lib87c\lvgl\extra\widgets\win\lv_win.c.o
In file included from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/meter/../../../../src/misc/lv_log.h:16,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/meter/../../../../lvgl.h:25,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/meter/../../../lvgl.h:17,
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/meter/lv_meter.h:16,    
                 from .pio/libdeps/esp32dev/lvgl/src/extra/widgets/meter/lv_meter.c:9:     
.pio/libdeps/esp32dev/lvgl/src/extra/widgets/meter/../../../../src/misc/../lv_conf_internal.h:46:17: note: #pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors
[... cut ...]
c:/users/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\src\main.cpp.o:(.literal._Z10mainscreenv+0x8): undefined reference to `lv_font_montserrat_22'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
=============================== [FAILED] Took 15.51 seconds ===============================
 *  The terminal process "C:\Users\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
`

rzeldent, 2023-03-06:
Hi Petr,

There is a default lv_conf.h present in the lvgl library, as a template.
This has to be adapter to what you need in your project, with the same name.
To do this there is an include directive present in the platformio.ini file
that points to the correct file, (should be the modified).
Unfortunately this location has to be absolute in platformio.h.

Hope this gives some more insight...

Kind regards

Rene

petr-vytlacil, 2023-03-06:
I'm sorry, it's my mistake. I forgot upload the config to the folder lvgl .-)
Sorry and THX

rzeldent, 2023-03-06:
You're welcome, things like this happen...

Enjoy!

bob-u, 2023-04-22:
I also struggled to get the lv_conf.h to be found on Windows.  Ultimately I had to put it right in the lvgl folder (not 'next' to it like template comments suggest).

-D LV_CONF_PATH=${platformio.include_dir}/lv_conf.h does not result in proper path (lv_conf.h in include folder) on windows for some reason.

MkMunich, 2023-08-25:
Hi, I got it compiled & built by renaming directory esp32-smartdisplay\examples\lvgl_pushbutton to \esp32-smartdisplay\examples\src and the following platformio.ini content:

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

monitor_speed = 115200

lib_ldf_mode=deep

build_flags =
    -Ofast
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    # LVGL settings
    -D LV_CONF_PATH="lv_conf.h"
    # Smartdisplay settings
    -D TFT_PANEL_ORDER_RGB
    #-D TFT_PANEL_ORDER_BGR
    #-D TFT_ORIENTATION_PORTRAIT
    -D TFT_ORIENTATION_LANDSCAPE
    #-D TFT_ORIENTATION_PORTRAIT_INV
    #-D TFT_ORIENTATION_LANDSCAPE_INV
    #-D ESP32_2432S028R
    #-D ESP32_3248S035R
    -D ESP32_3248S035C
    -I ../src
    -I ../include
    -I ../examples/src
	-I ".pio\libdeps\esp32dev\GFX Library for Arduino\src"
lib_deps = 
	Adafruit GFX Library@^1.11.7
	rzeldent/esp32_smartdisplay@^1.0.6
	lvgl/lvgl@^8.3.9

closed 2023-11-16 by rzeldent