rzeldent/esp32-smartdisplay #10: -D TFT_ORIENTATION_PORTRAIT_INV only changes touch orientation but not tft

NiklasVoigt, 2023-03-10:
I am using the ESP32_3248S035C.
As the title says, the build flag TFT_ORIENTATION_PORTRAIT_INV only changes the touch orientation but not the tft orientation.

build_flags =
    # LVGL settings
    -D LV_CONF_PATH=${platformio.include_dir}/lv_conf.h
    # Smartdisplay settings
    -D TFT_PANEL_ORDER_RGB
    -D TFT_ORIENTATION_PORTRAIT_INV
    -D ESP32_3248S035C

rzeldent, 2023-03-11:
Hi Niklas,

I looked at the code but do not see a reason why this works differently than on my board.
Is this still a problem?

NiklasVoigt, 2023-03-12:
Yes, this is still a problem.

rzeldent, 2023-03-12:
Hi Niklas,

Found the issue. Please check the branch at
https://github.com/rzeldent/esp32-smartdisplay/tree/bugfix/display_orientation
Let me know if this works for you...

Rene

rzeldent, 2023-03-12:
Created a test project https://github.com/rzeldent/esp32-smartdisplay-demo and tested the. orientation.
Error was because of incorrect ifdefs. Should be fixed now and is already merged into the main branch.
This will become version 1.0.5

Thanks for mentioning this bug!

NiklasVoigt, 2023-03-22:
Hi Rene, fixed!

closed 2023-03-22 by NiklasVoigt