rzeldent/esp32-smartdisplay #13: Display update question

mihai-manea, 2023-04-07:
Hi Rene,

I have one question. Do you know if the behavior that you see in the video is due to hardware limitations of the board or is something linked to software(library or something else). When I change the screen, it takes a while and if I add more things on the second screen, it freezes. If I manage to change the screen, then the widgets are updating fast.

https://youtu.be/-ZpO9WlQyLE

rzeldent, 2023-04-07:
Hi Mihai,

Maybe related to LVGL? Maybe the former screen is still in memory or needs to be rendered? Maybe double buffering?

Updates are done using SPI so should be quite fast. The bus speed can be tweaked but is already quite high (may be too high?)

mihai-manea, 2023-04-07:
[Squareline.zip](https://github.com/rzeldent/esp32-smartdisplay/files/11178597/Squareline.zip)

Here is my code. Can you please have a look because I don't know what to do. The change of SPI frequency doesn't do anything.

rzeldent, 2023-08-28:
Hi mihai,

The only thing I can think of is the update buffer. This is a small buffer to only send updates of the screen. This would explain why the widgets are updated quickly.
However, when switching screens all the changes will have to pass through this buffer. You might try to enlarge the buffer that is currently set at 10 times the width.

  // Setup TFT display
  lvgl_tft_init();
  static lv_disp_draw_buf_t draw_buf;
  static lv_color_t buf[TFT_WIDTH * 10];
  lv_disp_draw_buf_init(&draw_buf, buf, NULL, TFT_WIDTH * 10);

rzeldent, 2023-08-28:
I made a new option in the [esp32_smartdisplay.h](https://github.com/rzeldent/esp32-smartdisplay/blob/main/include/esp32_smartdisplay.h) to change the buffersize. The setting is called DRAW_BUFFER_SIZE

closed 2023-08-28 by rzeldent

mihai-manea, 2023-08-31:
Thank you,

It helped. Screen changing is much faster now. I used x50.
Aniway, it is not very smooth but I can live with this.