rzeldent/esp32-smartdisplay #14: Extension ports information

erich74, 2023-04-08:
Hi Rene,

Sorry this is not really an issue but I would like to get more info about the extension ports on this board.
There are 2 extensions called ' Extended IO' and 'Humidity interface'. I would like to understand if they are the same and if this is safe using them with I2C or SPI devices? Is it ok to use both at the same time?

Also I do not understand how to use the 'Power supply base'. Is it meant to supply the board with a 1s lipo or 18650 battery? If so what are the 2 additionnal pins for? does the board have a charging (protected?) circuit?

If you have any documentation answering my questions I would be happy to read through it.

Thanks a lot for your answers!
Eric

rzeldent, 2023-04-14:
Hi Eric,

I found an anotated schematic from MacsBugs: [esp32-3248s035r_pcb_layout-5](https://macsbug.files.wordpress.com/2022/09/esp32-3248s035r_pcb_layout_580s-4.jpg?w=580&zoom=2)

Here you can see that the connectors are connected to GPIO pins so you can use them how you want.

The power supply connector also has outputs of TxD2 and RxD2 so another serial output.

Think the document above (and maybe the site) will answer your questions.

Regards,

Rene

erich74, 2023-04-15:
Thank you very much Rene, I guess this schematic will answer all my questions!

closed 2023-04-15 by erich74

sartyx, 2023-12-26:
I wonder if someone was able to successfully power the newer ESP32-2432S024 by using the battery connector.
To safe some space I tried this today but it doesn't really work. It just gives the onboard LED a bit power but everything else stays "off". The documentation about it is almost useless unless you get deeper into the circuit.
It doesn't even mention if it works with 3.7V just:

"Supports lithium battery power supply, supports both charging and
discharging, and has over charging and over discharging protection."

I tried with some 5V power bank which I validated on the USB-C port.

Obviously there is some sort of a battery management on board but who knows how it works or what the trigger is actually doing.

bobskigit, 2023-12-27:
I have it working successfully with a LiPo battery . Picture enclosed
[image]

sartyx, 2023-12-27:
Ah ok, if I understand the specs,  there already is a battery management on board? So the LiPo is charging once connected to USB?

Gotta find some unused used Li* to try.