rzeldent/esp32-smartdisplay #15: Would you consider adding this model to your driver list

lesintheusa, 2023-04-30:
https://www.aliexpress.us/item/3256804766411337.html
The main difference is that it is an s3 board.

rzeldent, 2023-08-28:
Hi Lesintheusa,

I create a branch https://github.com/rzeldent/esp32-smartdisplay/tree/feature/esp32_8048S070 for the new type.
As I do not own the board I cannot define the calibration constants or check that it is working.

Maybe you (or someone else) can give it a try?

rzeldent, 2023-09-25:
Should be working. Let me know if there are any problems...

closed 2023-09-25 by rzeldent

joined, 2023-09-25:
@rzeldent I think that board uses a sort of parallel interface, not SPI, so I have doubts it would work with this library :sweat_smile: 

See e.g. https://forum.squareline.io/t/howto-esp32-8048s070-board-for-arduino/1098 and https://github.com/MMlodzinski/Sunton_ESP32-8048S070c_ESP_IDF_LVGL_example/tree/main

joined, 2023-09-25:
Another interesting link https://wiki.makerfabs.com/Sunton_ESP32_S3_5_7_inch_IPS_with_Touch.html