rzeldent/esp32-smartdisplay #19: PIO 3.3.0 ${platformio.src_include} needs to be ${PROJECT_INCLUDE_DIR}

XDGFX, 2023-07-16:
When building [the example](https://github.com/rzeldent/esp32-smartdisplay-demo) I was getting include errors when using the provided `platformio.ini`. For some reason the include directory was being provided without path separators, i.e. for some reason I was getting the error:
```
C:UsersUserProjectDirinclude/lv_conf.h: No such file or directory
```

Replacing `${platformio.src_include}` with `${PROJECT_INCLUDE_DIR}` (taken from [here](https://github.com/platformio/platformio-core/blob/4839fe37a3d1433737c80147ce9a205100dc4925/platformio/builder/main.py#L110-L129)) fixed the issue.

rzeldent, 2023-07-16:
Hi Callum,

I'll take a look what's going on!

Thanks sharing for the info; it might also help others facing this issue!

Rene

marcusobrien, 2023-07-25:
> When building [the example](https://github.com/rzeldent/esp32-smartdisplay-demo) I was getting include errors when using the provided `platformio.ini`. For some reason the include directory was [...]

Awesome, thanks for this info, saved me an hour messing about looking for how to do this, then forcing it with a static fixed path !

rzeldent, 2023-08-07:
Hi Marcus,

I put your fix in the repository. Thanks for resolving this!

closed 2023-08-07 by rzeldent