rzeldent/esp32-smartdisplay #27: ESP32-2432S024

busbuyukefe, 2023-09-23:
Hi, there is a new variant I found on AliExpress with a 2.4" ili9341 based lcd with capacitive and resistive touchscreen options, a powerbank IC (IP5306) for using it with lithium batteries and an USB Type C port for programming.

https://www.aliexpress.com/item/1005005865107357.html

The only problem is the capacitive version has the CST820 driver chip, I couldn't find the datasheet for the chip but I was able to find an arduino sketch that can read touch points from the touchscreen. 

https://gist.github.com/busbuyukefe/4a17159627c3ea83b4994f346002f061

I can confirm this code works for the touchscreen.

I tried to create a wrapper based on your library but I couldn't manage to get it done, do you think it is possible to add CST820 functionality without much work?

rzeldent, 2023-09-23:
Hi Busybuyukefe,

The link is to the original board? Or is it the board but with another capacitive controller chip?
This should be possible... Code does not look very complicated...

busbuyukefe, 2023-09-23:
Yes it is the link to the board ESP32-2432S024, they sell it in 3 variants (capacitive, resisitve, and no touchscreen) I bought a few of the capacitive touch versions and they have CST820 touch controller ic on them.
[image]
[image]

rzeldent, 2023-09-23:
Looks like a new revision of the 2432S035 board, let's hope only the controller has changed...

busbuyukefe, 2023-09-23:
> Looks like a new revision of the 2432S035 board, let's hope only the controller has changed...

Yes I can confirm only the touch controller (and maybe the backlight pwm pin) is changed. I was able to get display output with your demo.

rzeldent, 2023-09-23:
Hi,

Made a new branch with those boards. Untested as I do not own them.
Maybe you can give it a try?

rzeldent, 2023-09-24:
Nice!!!

I updated the branch with your findings; the spi_ili9431 was a typo/copy paste...
Gesture None was indeed something that I thought about...

Give it a try to see if this works, then I can merge it!

busbuyukefe, 2023-09-24:
There were a few changes that needed to be done, I created a PR  #12  for the needed changes, now It works without a problem thanks to your work :)

rzeldent, 2023-09-25:
Hi Busbuyuke,

Thanks for the PR and these fixes... Merged the branch so should be available!

closed 2023-09-25 by rzeldent