rzeldent/esp32-smartdisplay #32: how can i use this with squareline studio

TheRegularDX, 2023-10-06:
so i got my board (ESP32-2432S028R) and made a simple ui with squareline studio, the software made by lvgl to create draag and drop ui's, but i dont understand how to use the exported files from SQS on this library, is it possible?.

theres a lack of good tutorials and guides for SQS and im new using lvlg, and its my first time trying it

thank you.

rzeldent, 2023-10-07:
Hi RegularDX,

I took a look at squareline. Nice! It will create the c code for the UI.

From what I've seen and generated, it creates a file called ui.c that contains:

```
void ui_init( void )
{
lv_disp_t *dispp = lv_disp_get_default();
lv_theme_t *theme = lv_theme_default_init(dispp, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_RED), false, LV_FONT_DEFAULT);
lv_disp_set_theme(dispp, theme);
ui_Screen1_screen_init();
ui____initial_actions0 = lv_obj_create(NULL);
lv_disp_load_scr( ui_Screen1);
}
```

This file starts the gui after lvgl is initialized.

I think a project with just the esp32-smartdisplay and initializing the ui (ui_init()) will do!

rzeldent, 2023-10-07:
Hi RegularDX,

Played a lit more and adapted the demo application to be generated from [SquareLine Studio](https://squareline.io).
Might indeed be a bit easier and the personal license is free!

Enjoy!

TheRegularDX, 2023-10-07:
Thank you very much! Ill try that out