rzeldent/esp32-smartdisplay #34: Help

adonic, 2023-10-08:
Hello.
I’ll start by saying that I’m a newbie, I’m collecting information and trying to figure it out myself, but I understand that I can’t figure it out without the help of professionals. Now I am dealing with ESP32-3248S035S and from the first step I was faced with a lack of information and working demo options for studying and testing. I found your article, but I'm not familiar with PlatformIO.
I got acquainted with SQUARDLINE STUDIO, but was faced with the fact that it didn’t work from the first step and quit again because I didn’t know what the problem was. I have a request to YOU, help me make a working version of my project, based on it I will try to move forward in my training.
I can provide you with a GUI to create a working DEMO.
I'm studying on the basis of ArduinoIDE, but if you have time or have instructions for starting my project on PlarformIO
i will be very happy to accept this experience, i will study this direction too.

rzeldent, 2023-10-08:
Hi adonic,

I suggest installing PlatformIO and get a bit acquainted with it.
Then just open the Github repository: [smartdisplay-demo](https://github.com/rzeldent/esp32-smartdisplay-demo). This contains a demo using Squareline studio. Try to get it running and opening the SquareLine project file.

To create a working application, remove all files in the UI directory and let SquareLine studio export the UI files to this directory. This should be enough for a demo. I think you'll have to adapt the display size etc.

Good luck!

adonic, 2023-10-08:
Thank you very much for your answer, I’ve been figuring it out for the second day now, I’ll try as you said, if I have a question, can I ask you for help?

adonic, 2023-10-10:
Good afternoon. I'm frustrated, I can't work with the screen, I can't put together a working example. Maybe you have a little example ready because I built my screen in Squareline and then can't get it to work correctly with PlatformIO and then send it to my board.

closed 2023-10-10 by rzeldent