rzeldent/esp32-smartdisplay #38: ESP32-8048S043 with Capacitive Touch

BlueHaze1963, 2023-10-28:
Hi,

Great job on putting this together, so I was very hopeful when I started with this library. However, I have the 4.3inch version of the 800x480 display instead with a capacitive touch screen. There seems to be no board definition for this one, so I selected the ESP32_8048S070C, but the screen stays black. Is it possible to add this board definition or alternatively, any indications of what I should change in the include files to get this board to work ?

Thanks, Patrick

rzeldent, 2023-10-28:
Hi BlueHaze,

What a bummer. 
ESP32_8048S070C seems a good choice... However, if the screen stays black the initialization is not performed correctly.
Maybe another graphics controller or some bugs in the current implementation.
Do you have pictures of the board? What does it says for the board type?

BlueHaze1963, 2023-10-28:
Hi Rene, 
I did some searching and it turms out it is this board :  https://wiki.makerfabs.com/Sunton_ESP32_S3_4.3_inch_800x400_IPS_with_Touch.html
Using the example code on that wiki site, I was able to program the board from Arduino. 
It seems that although the 7" and 4.3" are quite similar, the difference seems to be in the pinassignments between the 2 displays.

These seem to do the trick : 
#define TFT_BL 2 // backlight pin

#define SD_SCK 12
#define SD_MISO 13
#define SD_MOSI 11
#define SD_CS 10

#define TOUCH_SCL 20
#define TOUCH_SDA 19
#define TOUCH_INT -1
#define TOUCH_RST 38
#define TOUCH_ROTATION ROTATION_NORMAL
#define TOUCH_MAP_X1 480
#define TOUCH_MAP_X2 0
#define TOUCH_MAP_Y1 272
#define TOUCH_MAP_Y2 0

Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
    40 /* DE */, 41 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    45 /* R0 */, 48 /* R1 */, 47 /* R2 */, 21 /* R3 */, 14 /* R4 */,
    5 /* G0 */, 6 /* G1 */, 7 /* G2 */, 15 /* G3 */, 16 /* G4 */, 4 /* G5 */,
    8 /* B0 */, 3 /* B1 */, 46 /* B2 */, 9 /* B3 */, 1 /* B4 */
);

// LCD 800x480
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
    bus,
    800 /* width */, 0 /* hsync_polarity */, 8 /* hsync_front_porch */, 4 /* hsync_pulse_width */, 8 /* hsync_back_porch */,
    480 /* height */, 0 /* vsync_polarity */, 8 /* vsync_front_porch */, 4 /* vsync_pulse_width */, 8 /* vsync_back_porch */,
    1 /* pclk_active_neg */, 16000000 /* prefer_speed */, true /* auto_flush */);

TAMC_GT911 ts = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, max(TOUCH_MAP_X1, TOUCH_MAP_X2), max(TOUCH_MAP_Y1, TOUCH_MAP_Y2));

Now trying to find out how/where to change or add this in your library

Kind regards, Patrick

rzeldent, 2023-10-29:
Hi Patrick,

The way of communicating has been changed, not SPI but something parallel. Also, looking at the schematic, the speaker output is no longer there. Will have to see how to get this working without a board to test...

BTW saw the video on the main page of MakerLabs and looks the drawing is very slow?!

rzeldent, 2023-10-29:
Ordered a similar board to check what's going on...
https://www.aliexpress.com/item/1005004788147691.htm

rzeldent, 2023-11-05:
Hi Patrick,

Got the board today and got it working. Is indeed completely different!

Still work in progress but take a look at https://github.com/rzeldent/esp32-smartdisplay-demo/tree/feature/esp32s3

Maybe you can give it a test?

BlueHaze1963, 2023-11-09:
Hi Rene,

Downloaded the new git repository, but not sure to change/define where to make the 8048S43C display work with your library.

Mvg,
Patrick

rzeldent, 2023-12-06:
HI Patrick,

The demo application download/uses the esp32-smartdisplay library. To run the demo and start from there, just uncomment the right board and (optionally) set the board variable in the [platformio] section to the processor in your board (esp32dev) so:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
#default_envs = esp32dev
#default_envs = esp32-c3-devkitm-1
default_envs = esp32-s3-devkitc-1

[env]
platform = espressif32
framework = arduino

monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_filters = esp32_exception_decoder

# Partition scheme for OTA
board_build.partitions = min_spiffs.csv

build_flags =
    -Ofast
    #-Wall
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    # LVGL settings
    -D LV_CONF_PATH=${PROJECT_INCLUDE_DIR}/lv_conf.h
    # EspressIf library defines
    -D ESP_LCD_ST7796_VER_MAJOR=1
    -D ESP_LCD_ST7796_VER_MINOR=2
    -D ESP_LCD_ST7796_VER_PATCH=0
    -D ESP_LCD_ILI9341_VER_MAJOR=1
    -D ESP_LCD_ILI9341_VER_MINOR=2
    -D ESP_LCD_ILI9341_VER_PATCH=0
    -D ESP_LCD_GC9A01_VER_MAJOR=1
    -D ESP_LCD_GC9A01_VER_MINOR=2
    -D ESP_LCD_GC9A01_VER_PATCH=0
    -D CONFIG_ESP_LCD_TOUCH_MAX_POINTS=1
    -D CONFIG_XPT2046_CONVERT_ADC_TO_COORDS
    -D CONFIG_XPT2046_Z_THRESHOLD=600

#lib_deps = rzeldent/esp32_smartdisplay
lib_deps =
    https://github.com/rzeldent/esp32-smartdisplay.git#feature/esp32s3
    https://github.com/schreibfaul1/ESP32-audioI2S

[env:esp32dev]
board = esp32dev

build_flags =
    ${env.build_flags}
    # Smartdisplay selection
    #-D ESP32_2432S024N
    #-D ESP32_2432S024R
    #-D ESP32_2432S024C
    #-D ESP32_2432S028R
    #-D ESP32_2432S032N
    #-D ESP32_2432S032R
    #-D ESP32_2432S032C
    #-D ESP32_3248S035R
    #-D ESP32_3248S035C

lib_deps = ${env.lib_deps}

[env:esp32-c3-devkitm-1]
board = esp32-c3-devkitm-1

build_flags =
    ${env.build_flags}
    # Smartdisplay selection
    #-D ESP32_2424S012N
    #-D ESP32_2424S012C

lib_deps = ${env.lib_deps}

[env:esp32-s3-devkitc-1]
board = esp32-s3-devkitc-1

build_flags =
    ${env.build_flags}
    # Smartdisplay selection
    #-D ESP32_1732S019N
    #-D ESP32_1732S019C
    #-D ESP32_4827S043N
    #-D ESP32_4827S043R
    #-D ESP32_4827S043C
    #-D ESP32_8048S043N
    #-D ESP32_8048S043R
    -D ESP32_8048S043C
    #-D ESP32_8048S050N
    #-D ESP32_8048S050R
    #-D ESP32_8048S050C
    #-D ESP32_8048S070N
    #-D ESP32_8048S070R
    #-D ESP32_8048S070C

lib_deps = ${env.lib_deps}
```

Notice ESP32_8048S043C has been uncommented

BlueHaze1963, 2023-12-09:
Hi Rene,

Downloaded the feature version, used the platformio file you specify below and tried to compile, but get several compilation errors

/src/LvglWidgets.ino:43:1: error: no matching function for call to 'Arduino_ESP32RGBPanel::Arduino_ESP32RGBPanel(int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int, int,
int, int, int, int)'

src/LvglWidgets.ino:46:1: error: 'Arduino_RPi_DPI_RGBPanel' does not name a type; did you mean 'Arduino_ESP32RGBPanel'?

src/touch.h:163:71: error: 'gfx' was not declared in this scope
(3 times this error: 2nd time on src/LvglWidgets.ino:70:3 and 3rd time on src/LvglWidgets.ino:108:3)

src/LvglWidgets.ino:155:5: error: 'lv_demo_widgets' was not declared in this scope

What did I miss ?

Mvg,
Patrick

; https://docs.platformio.org/page/projectconf.html

    https://github.com/rzeldent/esp32-smartdisplay.git#feature/esp32s3
    https://github.com/schreibfaul1/ESP32-audioI2S

rzeldent, 2023-12-09:
Hi Patrick,

I do not know what you're compiling but /src/LvglWidgets.ino is not something that sounds familiar. This library is for PlatformIo.
gfx is something that is used by other libraries like TFT_eSPI. I suspect someting is mixed up.

BlueHaze1963, 2023-12-09:
I simply downloaded and extracted your feature branch. Opened the platformio file under example and uncommented
 -D ESP32_8048S043C
and hit the compile button

Probably I skipped a few important steps ? 😉

Mvg,
Patrick

rzeldent, 2023-12-09:
Hi Patrixk,

It is a library that is supposed to included with your program. Try to
clone: rzeldent/esp32-smartdisplay-demo at feature/esp32s3
(github.com)

BlueHaze1963, 2023-12-09:
Maybe I misunderstand waht you mean, but I think I did exactly that : download the feature branch, uncomment the 8048S043C line and compiled

[cid:41da8441-7723-4dfd-a053-947a979a9306]
then I got the errormessages

Mvg,
Patrick

BlueHaze1963, 2023-12-09:
Hi Rene,

Sorry for my silly questions, but I am quite new to platformio. So, it seems I had the wrong environment selected. I couldn't select your project as enviroment until I copied platformio.ini from the example folder to one level higher (under esp32-smartdisplay-feature-esp32s3) and then I could select "env:edp32-s3-devkit-1 (esp32-smartdisplay-feature-esp32s3)" to compile.

I also moved main.cpp to src  and lv_conf.h to the include folder

Now I get : "In file included from src/esp32_smartdisplay.c:1
include/esp32_smartdisplay.h:5:10: fatal error: lvgl.h: No such file or directory"

Any suggestions ?

Mvg,
Patrick

BlueHaze1963, 2023-12-09:
Hi Rene,

I cleaned the project, removed .pio and .vscode and this triggered it to reload the dependancies. If actually downloaded lvgl and some other libraries. I then recompiled and it went very well until I hit this (new) issue

.pio/libdeps/esp32-s3-devkitc-1/ESP32-audioI2S-master/src/Audio.h:17:10: fatal error: SPI.h: No such file or directory

Any suggestions ?(full compiler output attached)

Mvg,
Patrick

Processing esp32-s3-devkitc-1 (board: esp32-s3-devkitc-1; platform: espressif32; framework: 
arduino)
--------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.3.2) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32_smartdisplay @ 2.0.0+sha.b196771
|-- ESP32-audioI2S-master @ 2.0.0+sha.71b7523
|-- lvgl @ 8.3.9
Building in release mode
Compiling .pio\build\esp32-s3-devkitc-1\src\esp32_smartdisplay.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\esp_lcd_touch.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\esp_lcd_touch_cst816s.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\esp_lcd_touch_gt911.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\esp_lcd_touch_xpt2046.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\main.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\src\smartdisplay_cst816s.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\smartdisplay_gc9a01.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\smartdisplay_gt911.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\smartdisplay_ili9341.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\smartdisplay_rgb.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\smartdisplay_st7789.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\smartdisplay_st7796.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\smartdisplay_xpt2046.c.o
Building .pio\build\esp32-s3-devkitc-1\bootloader.bin
Generating partitions .pio\build\esp32-s3-devkitc-1\partitions.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_disp.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_event.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_group.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_indev.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_indev_scroll.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_obj.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_obj_class.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_obj_draw.c.o
.pio/libdeps/esp32-s3-devkitc-1/lvgl/src/core/lv_obj.c: In function 'lv_init':
.pio/libdeps/esp32-s3-devkitc-1/lvgl/src/core/lv_obj.c:152:18: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     char * txt = "Á";
                  ^~~~
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_obj_pos.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_obj_scroll.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_obj_style.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_obj_style_gen.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_obj_tree.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib45c\lvgl\core\lv_refr.c.o
[... 199 lines cut ...]
Compiling .pio\build\esp32-s3-devkitc-1\lib923\ESP32-audioI2S-master\flac_decoder\flac_decoder.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\lib923\ESP32-audioI2S-master\mp3_decoder\mp3_decoder.cpp.o
In file included from .pio/libdeps/esp32-s3-devkitc-1/ESP32-audioI2S-master/src/Audio.cpp:13:
.pio/libdeps/esp32-s3-devkitc-1/ESP32-audioI2S-master/src/Audio.h:17:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

 #include <SPI.h>
          ^~~~~~~
compilation terminated.
*** [.pio\build\esp32-s3-devkitc-1\lib923\ESP32-audioI2S-master\Audio.cpp.o] Error 1
=============================== [FAILED] Took 212.03 seconds ===============================
Environment         Status    Duration
------------------  --------  ------------
esp32-s3-devkitc-1  FAILED    00:03:32.028
========================== 1 failed, 0 succeeded in 00:03:32.028 ========================== 

 *  The terminal process "C:\Users\Patrick\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32-s3-devkitc-1'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

rzeldent, 2023-12-09:
Yes, you forgot to update the platform see differences in your output vs mine

**PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)**
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash

Run in the platformIO CLI:

PS C:\Users\rzeld\source\repos\esp32-smartdisplay-demo> pio pkg update     
Resolving esp32dev dependencies...
Already up-to-date.
Resolving esp32-c3-devkitm-1 dependencies...
Already up-to-date.
Resolving esp32-s3-devkitc-1 dependencies...
Already up-to-date.

closed 2023-12-09 by rzeldent

BlueHaze1963, 2023-12-09:
Thanks Rene.
The project now compiles successfully and I can upload it, but the screen remains black, even after a powercycle. Strange, as it worked fine with the preloaded demo program before I flashed it

Mvg,
Patrick

rzeldent, 2023-12-09:
Hi Patrick,

I must confess I do not own the ESP32-8048S043 so testing this board is a bit difficult. The preloaded program can be restored, they provided some instructions for that.... but the point is you can make your own display, right!

Does the light turns on or does the complete LCD stays dark?

rzeldent, 2023-12-09:
Hi Patrick,

I compiled it and tested it on the 4827S043R with the settings for the 8048S070. Indeed there is a memory warning and stops.
Think some memory is allocated by the driver. Will investigate...

rzeldent, 2023-12-09:
Hi Patrick,

I checked around and think it has something to do with the PSRAM. I myself own a ESP32-4827S043R. This works perfectly with the 16 bits parallel DMA connection.

However, if I change to the ESP32-8048S070, i get:

 152][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!
[   414][I][main.cpp:27] setup(): CPU: ESP32-S3 rev0, CPU Freq: 240 Mhz, 2 core(s)
[   414][I][main.cpp:28] setup(): Free heap: 315260 bytes
[   416][I][main.cpp:29] setup(): SDK version: v4.4.5
E (422) lcd_panel.rgb: esp_lcd_new_rgb_panel(151): no mem for frame buffer

What can be seen is that the memory available is only 315260 bytes at startup.

For 480x272x2 in total 261120 bytes are required, leaving only 54140 bytes ram for the application. But for the 8048S070, in total 800x480x2 = 768000 bytes are required. This is more than the RAM so the message is correct.

I suspect the higher resolutions use the PSRAM for the display buffer. I noticed that 8Mb PSRAM was mentioned in the description. I made some modifications in an attempt to support PSRAM but am unable to test.
In the meantime I ordered an ESP32-8048S043C, another $40 :-( to test this. So a bit of patience...
I'll check in the modifications so far, maybe it is enough to get it working but have doubts...

BlueHaze1963, 2023-12-10:
Re-downloaded the repository, compiled & uploaded it to the board, but no luck. Still a black screen

Mvg,
Patrick

rzeldent, 2023-12-10:
Thanks for checking. Will have to wait till it arrives...