rzeldent/esp32-smartdisplay #39: How to start, please?

cody-assistant, 2023-11-01:
Hi.
Im new with this device and after lot of hours rounding around, im unable to understand how to make run this. Can you, please, give me the first steps about how to setup the folders and how to start the project? thanks very much in advance

rzeldent, 2023-11-13:
Hi cody,

Use the esp32-smartdemo project to get started. This is a demo application that uses this library.
See the documentation: https://github.com/rzeldent/esp32-smartdisplay-demo

cody-assistant, 2023-11-15:
Thanks for your response, but the link you post is for python version and also: 
"See [esp32-smartdisplay](https://github.com/rzeldent/esp32-smartdisplay) for more information"
¿?

cody-assistant, 2023-11-15:
Thanks for your response. i got it running in ESP32_2432S024C device.
how to run in a ESP32-S3-WROOM-1 device? what i need to change?
thanks in advance

cody-assistant, 2023-11-15:
How to create a QR Code for ESP32_2432S024C?

rzeldent, 2023-11-15:
Hi Cody,

You can just change the value for the board=  in the platformio.ini
fille to match your board!
However, this repo is specially for the Sunton boards. If you're interested
in the latest boards (with esp32-s3) take a look at the feature branch!

Rene

cody-assistant, 2023-11-15:
Hi and thanks very much for your help.
I have downloaded feature branch and tried to compile. After moving files from "example" folder to root foder and copy some .h files, i get some errors and i can not continue with compilation:

.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: 
.pio/build/esp32dev/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'

.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: 
.pio/build/esp32dev/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'

.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: 
.pio/build/esp32dev/libFrameworkArduino.a(main.cpp.o): in function `loopTask(void*)':

Can you, please, give me some help?
Thanks in advance

cody-assistant, 2023-11-15:
> Hi cody,
> 
> Use the esp32-smartdemo project to get started. This is a demo application that uses this library. See the documentation: https://github.com/rzeldent/esp32-smartdisplay-demo

This is working perfectly. I have a question. Is it possible to add to the project a TCP server accepting data from connection to a given port and show the data received while the current program in esp32-smartdisplay-demo continues working? 
Thanks in advance.

cody-assistant, 2023-11-15:
> Use the esp32-smartdemo project to get started. This is a demo application that uses this library.
> See the documentation: https://github.com/rzeldent/esp32-smartdisplay-demo

With this, i have not the board im using. The board is: ESP32- 8048S043, and this board is present in the platformio.ini in the feature branch.

rzeldent, 2023-11-16:
Hi cody,

It is present but it still under development, just remove the # in front of ESP32_8048S043. N=no touch, R=resitive touch and C=capacitive touch,

rzeldent, 2024-01-01:
ESP32- 8048S043 is working

closed 2024-01-01 by rzeldent