rzeldent/esp32-smartdisplay #40: ESP32 8048S070C

Dave3891, 2023-11-05:
Has anyone got the 8048S070C working? I tried the bare minimum branch with the -D ESP32_8048S070C flag set and the board set to board = esp32-s3-devkitc-1. 
It compiles and uploads but nothing is displayed. I also tried adding these settings, but still nothing.
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.arduino.partitions = default_16MB.csv
board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_upload.flash_size = 16MB

rzeldent, 2023-11-05:
Hi Dave,

See issue https://github.com/rzeldent/esp32-smartdisplay/issues/40

closed 2023-11-05 by rzeldent

Dave3891, 2023-11-05:
But #40 is this issue

Dave3891, 2023-11-05:
> Hi Dave,
> 
> See issue #40

I am assuming you mean issue #38 
I just tried the code from https://github.com/rzeldent/esp32-smartdisplay-demo/tree/feature/esp32s3
But it is still just a black screen with no backlight or anything. Any ideas?

rzeldent, 2023-11-05:
There is also a definition present for the 8048s070c. I got the 4827
version working using the same hardware