rzeldent/esp32-smartdisplay #41: get_light_intensity() is always returning zero

fcortijos, 2023-11-17:
First of all, I would like to thank you for your efforts in creating this library.
I have the ESP32-3248S035C module. The function smartdisplay_get_light_intensity() is always returning zero.
In the esp32_smartdisplay.h file the CDS_PIN is pointed to pin 34.
Is there any particular pin configuration for this board or I bought  a defective one?

tks

rzeldent, 2023-11-17:
Hi fcortijos ,

Normally, with light the value is 0. However, when covered the value increases.
So in a way, you can call it a dark-sensor.

Please try if this is correct!

Regards

fcortijos, 2023-11-21:
Hi, it didn't change the value.

So I use this code instead (bellow) and the value change from 75 (no cover) to 400 (when covered).

  // lv_label_set_text(label_cds, String(smartdisplay_get_light_intensity()).c_str());
  lv_label_set_text(label_cds, String(analogReadMilliVolts(LIGHT_SENSOR)).c_str());

Regards

lazzaroevan, 2023-12-29:
I have a problem with my light sensor, it seems to only be set to 75, and if i completely cover it, nothing changes, pretty sure its set to the right gpio pin as well.

rzeldent, 2024-02-02:
> I have a problem with my light sensor, it seems to only be set to 75, and if i completely cover it, nothing changes, pretty sure its set to the right gpio pin as well.

Have you checked the demo? It is displayed on the screen

lazzaroevan, 2024-02-05:
Yes, while running the demo, the value was stuck at 75, on one of my other boards, when I specifically set the dimming it seemed to work properly, so it may be a bad board or two.

rzeldent, 2024-02-18:
Then it is well possible something is wrong with the CdS sensor or soldering.
When the sensor is in the dark, it has a high impedance so the value should be high. However in your case it looks like it is shorted because the value is always 75. Maybe some solder is present below the CdS sensor?
You could try to take them out and retry soldering...