rzeldent/esp32-smartdisplay #42: Demo just flashing the LED

sartyx, 2023-11-19:
Hello,

am I missing smth or is the demo just flashing the LED on the display board? I expected something fancy on the display ;-)
Took me a while to figure out that this one doesn't work with the arduino IDE *sigh* so finally I got it into platformIO 
but once I uploaded it it is doing nothing but the blinking LED. 

using the non touch version: -D ESP32_2432S024N

regards

rzeldent, 2023-11-19:
Hi Sartyx,

Good you got it compiled with PlatformIO, one hurdle passed already!

I cannot tell you from there why but you can take a look at the logging. Maybe no backlight?
Even better, try the soon to be released version on the [feature/esp32s3](https://github.com/rzeldent/esp32-smartdisplay/tree/feature/esp32s3) branch. This uses new drivers and also supports the newer boards.

I ordered a 2432S24 because I got the parameters from the schematic (and had the others running). Will let you known what happend when I compiled it...

sartyx, 2023-11-20:
I forced some backlight 
    smartdisplay_tft_set_backlight(512);

no change... added
    smartdisplay_tft_wake();

nothing

guess I will try the branch now

rzeldent, 2023-11-21:
Reproduced. Working on it!

rzeldent, 2023-11-21:
Fixed, was a typo. Used new updated (will be the default soon) https://github.com/rzeldent/esp32-smartdisplay.git](https://github.com/rzeldent/esp32-smartdisplay/tree/feature/esp32s3)

Demo using https://github.com/rzeldent/esp32-smartdisplay-demo:

[image]

rzeldent, 2023-11-21:
Still working on esp32-2432024R

rzeldent, 2023-12-06:
And?

sartyx, 2023-12-09:
I have no idea why but platformio hates me. I need to get i working in arduino ide somehow, I simply don't get the option to open anything in platformio as described in their quickstart guide, there is no more "Home" or alternative link to let me work with it in visual studio. Already deinstalled and reinstalled it. No clue.

sartyx, 2023-12-09:
Ok got it working, finally.
Not as seen on PXL_20231130_225143662.jpg  though, just a blank screen with a button and a milliseconds counter.
Gotta find out what I can do now. Never worked with Squareline Studio, doubt I will need it. 
I already found out that the few available GPIO pins (35,22) can be used for a HC-12 433Mhz radio communication, which is neat because it allows a more distant distance communication than BT or Wifi.

rzeldent, 2023-12-10:
Pfff... nice to hear you got it working. :thumbsup:

Give SquareLine Studio a try. It takes a liitle time but then it is really easy to add screens/transitions etc...
Best part is that the UI code is generated for you!

closed 2023-12-10 by rzeldent

MikeEitel, 2024-03-17:
Just FYI
I have a solution, working on all three variants N / C / R.
Lot's of misinformation is out about this devices. The pin numbering is different, C uses another chip, R has must used patch to HSPI ...... In the moment I found that SDA does not work as the have a connection UNDER CN1 to ground... Grrrr

rzeldent, 2024-03-17:
Hi Mike,

Interesting?! Are they different that the current main? I do have my esp32-2432S022C working!

MikeEitel, 2024-03-17:
Hi Rene

I tried to attach a zip archive ( send me your direct contact and I can send it
with my current work in process. ( I the moment I apply I2C senors ) 
( I'm ready to do this as I appreciated your work very much when I was fiddling around with this implementation )

You decide in platformio.ini the type of board you want to use.

I found that the IO pin numbering is different depending of the boards ( you have all in the .ino)

The resistive is even using a newer chip. The capacitive has the  problem of using the same HSPI but with two speeds (60 / 2Mhz)

I do not know why but (at least N and R version) has a resistor R25 with 0 ohm that shorts the SDA pin… (Grrrrr, cost’s me days)
Now I can use the pins of CN1 for SDA and SCL. So in the moment I’m in work to implement different I2C sensors. 

And yes. This device is running MQTT. ( I have a suit of devices for my home, based on docker, mosquito, NodeRed … [SensorsIOT] )
It is enough to setup a mqtt server so the device starts. When not connected to wifi and mqtt it will always restart. But
probably you are able to patch that. Commands given via mqtt explorer are good enough to command it, examples are in the comments.

And it is running both cpus. 
I was not able to split screen, touch and mqtt to different tasks, I had glitches on the screen. But the sensors run successfully on cpu0.

Hope you can find some info in my code that helps you further on.

P.S.
No I’m not a real coder and also have no intention to become one. I’m just an retired engineer who wants to have a solution.
Eventually I will download to my (empty) github, but do not intend to react on issues… Sorry

Best regards 
Mike Eitel  
[mike    at    eitel     ch]

sartyx, 2024-03-18:
Are we talking about the same boards? Do you apply the I²C sensors directly? I haven't inspected the circuit closely but are the few open GPIO actually SDA and SCL? The specs of board don't offer I²C, but the chip would of course.

MikeEitel, 2024-03-18:
I have all three variants working of:
https://de.aliexpress.com/item/1005005865107357.html?pdp_npi=4%40dis%21CHF%21CHF%207.06%21CHF%206.50%21%21%2156.43%2151.92%21%40210384b217107486678332027efe13%2112000034624084331%21sh%21CH%21138439492%21&spm=a2g0o.store_pc_allItems_or_groupList.new_all_items_2007567266044.1005005865107357&gatewayAdapt=glo2deu
And without the R25 (until now only desoldert on R type) they work with sda scl via CN1 connector.. I see no reason for R25. Eventually there will be a collision with pin numbers on C type.