rzeldent/esp32-smartdisplay #44: ESP32_8048S070C 'ESP_LCD_GC9A01_VER_MAJOR' undeclared

sag3ll0, 2023-12-04:
I have an [ESP32_8048S070C ](https://aliexpress.ru/item/1005005612035811.html)
also [info](https://esp3d.io/esp3d-tft/v1.x/hardware/sunton-7-8048/index.html)
setup board from the readme
```ini
[env:esp32-s3-devkitc-1]
board = esp32-s3-devkitc-1
board_build.arduino.memory_type = qio_opi
board_upload.flash_size = 16MB
board_build.arduino.partitions = app3M_fat9M_16MB.csv

lib_deps = 
    https://github.com/rzeldent/esp32-smartdisplay.git#feature/esp32s3	

build_flags =
    -Ofast
    -Wall
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    -D LV_CONF_PATH=${PROJECT_INCLUDE_DIR}/lv_conf_new.h
    -D ESP_LCD_ST7796_VER_MAJOR=1
    -D ESP_LCD_ST7796_VER_MINOR=2
    -D ESP_LCD_ST7796_VER_PATCH=0
    -D ESP_LCD_ILI9341_VER_MAJOR=1
    -D ESP_LCD_ILI9341_VER_MINOR=2
    -D ESP_LCD_ILI9341_VER_PATCH=0
    -D CONFIG_ESP_LCD_TOUCH_MAX_POINTS=1
    -D CONFIG_XPT2046_CONVERT_ADC_TO_COORDS
    -D CONFIG_XPT2046_Z_THRESHOLD=600
    -D ESP32_8048S070C
    -D BOARD_HAS_PSRAM     
```

when I try to compile a project, I get the following errors
```
.pio/libdeps/esp32-s3-devkitc-1/esp32_smartdisplay/src/esp_lcd_gc9a01.c:128:66: error: 'ESP_LCD_GC9A01_VER_MAJOR' undeclared (first use in this function); did you mean 'ESP_LCD_ILI9341_VER_MAJOR'?
.pio/libdeps/esp32-s3-devkitc-1/esp32_smartdisplay/src/esp_lcd_gc9a01.c:128:92: error: 'ESP_LCD_GC9A01_VER_MINOR' undeclared (first use in this function); did you mean 'ESP_LCD_ILI9341_VER_MINOR'?
.pio/libdeps/esp32-s3-devkitc-1/esp32_smartdisplay/src/esp_lcd_gc9a01.c:129:14: error: 'ESP_LCD_GC9A01_VER_PATCH' undeclared (first use in this function); did you mean 'ESP_LCD_ILI9341_VER_PATCH'?
```

if i set this values in build_flags its compiled ok

```ini
build_flags =
    -Ofast
    -Wall
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    -D LV_CONF_PATH=${PROJECT_INCLUDE_DIR}/lv_conf_new.h
    -D ESP_LCD_ST7796_VER_MAJOR=1
    -D ESP_LCD_ST7796_VER_MINOR=2
    -D ESP_LCD_ST7796_VER_PATCH=0
    -D ESP_LCD_ILI9341_VER_MAJOR=1
    -D ESP_LCD_ILI9341_VER_MINOR=2
    -D ESP_LCD_ILI9341_VER_PATCH=0
    -D ESP_LCD_GC9A01_VER_MAJOR=1
    -D ESP_LCD_GC9A01_VER_MINOR=2
    -D ESP_LCD_GC9A01_VER_PATCH=0
    -D CONFIG_ESP_LCD_TOUCH_MAX_POINTS=1
    -D CONFIG_XPT2046_CONVERT_ADC_TO_COORDS
    -D CONFIG_XPT2046_Z_THRESHOLD=600
    -D ESP32_8048S070C
    -D BOARD_HAS_PSRAM     
```

[image]

plate ESP32_8048S070C has version 1.3 and using EK9716 driver and capacitive touch panel - GT911 (i2C - 0x5D or 0x14)
Maybe I'm doing something wrong

sag3ll0, 2023-12-07:
problem is still present
version:

[image]

error:

[image]

rzeldent, 2023-12-07:
Hi Oleg,

You really need to update: ed9e2d5 is from dec 6th. You're at least 20 commits behind!

[image]

rzeldent, 2023-12-07:
Hi Oleg,

Yes but this is not the latest version!

rzeldent, 2023-12-09:
Hi Oleg,

I did check the latest version. I compiled the demo application.

Please take a look and look for any differences.

Opening the project
```
Resolving esp32-s3-devkitc-1 dependencies...
Library Manager: Installing git+https://github.com/rzeldent/esp32-smartdisplay.git#feature/esp32s3
git version 2.37.2.windows.2
Cloning into 'C:\Users\rzeld\.platformio\.cache\tmp\pkg-installing-l3ldf7z_'...

Library Manager: esp32_smartdisplay@2.0.0+sha.c2bc33e has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing lvgl/lvgl @ ^8.3.9
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: lvgl@8.3.9 has been installed!
Library Manager: Installing git+https://github.com/schreibfaul1/ESP32-audioI2S
git version 2.37.2.windows.2
Cloning into 'C:\Users\rzeld\.platformio\.cache\tmp\pkg-installing-jnnhrcut'...

Library Manager: ESP32-audioI2S-master@2.0.0+sha.71b7523 has been installed!
Updating metadata for the vscode IDE...
Project has been successfully updated!
```

Then the build:
```
 *  Executing task: C:\Users\rzeld\.platformio\penv\Scripts\platformio.exe run 

Processing esp32-s3-devkitc-1 (board: esp32-s3-devkitc-1; platform: espressif32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- esp32_smartdisplay @ 2.0.0+sha.c2bc33e
|-- ESP32-audioI2S-master @ 2.0.0+sha.71b7523
|-- lvgl @ 8.3.9
Building in release mode
Compiling .pio\build\esp32-s3-devkitc-1\src\main.cpp.o
Compiling .pio\build\esp32-s3-devkitc-1\src\ui\components\ui_comp_hook.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\ui\images\ui_img_lenna_165_png.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\ui\screens\ui_scrMain.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\ui\ui.c.o
Compiling .pio\build\esp32-s3-devkitc-1\src\ui\ui_helpers.c.o
Building .pio\build\esp32-s3-devkitc-1\bootloader.bin
Generating partitions .pio\build\esp32-s3-devkitc-1\partitions.bin
Compiling .pio\build\esp32-s3-devkitc-1\lib7e7\lvgl\core\lv_disp.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib7e7\lvgl\core\lv_event.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib7e7\lvgl\core\lv_group.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib7e7\lvgl\core\lv_indev.c.o
Compiling .pio\build\esp32-s3-devkitc-1\lib7e7\lvgl\core\lv_indev_scroll.c.o
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\esp32-s3-devkitc-1\lib7e7\lvgl\core\lv_obj.c.o
[... 291 lines cut ...]
Compiling .pio\build\esp32-s3-devkitc-1\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\esp32-s3-devkitc-1\lib74d\libESP32-audioI2S-master.a
Archiving .pio\build\esp32-s3-devkitc-1\libFrameworkArduino.a
Linking .pio\build\esp32-s3-devkitc-1\firmware.elf
Retrieving maximum program size .pio\build\esp32-s3-devkitc-1\firmware.elf
Checking size .pio\build\esp32-s3-devkitc-1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  21.3% (used 69764 bytes from 327680 bytes)
Flash: [===       ]  29.4% (used 578009 bytes from 1966080 bytes)
Building .pio\build\esp32-s3-devkitc-1\firmware.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
==================================================================================================== [SUCCESS] Took 58.10 seconds ====================================================================================================

Environment         Status    Duration
------------------  --------  ------------
esp32-s3-devkitc-1  SUCCESS   00:00:58.100
=============================================================================================
```

Hope this helps! Good luck.

Maybe try to do an update of the platform (this happened also to others) and clone the project again....