rzeldent/esp32-smartdisplay #45: ESP32_8048S070C  - no mem for frame buffer

sag3ll0, 2023-12-04:
I have an [ESP32_8048S070C ](https://aliexpress.ru/item/1005005612035811.html)
also [info](https://esp3d.io/esp3d-tft/v1.x/hardware/sunton-7-8048/index.html)
setup board from the readme
```ini
[env:esp32-s3-devkitc-1]
board = esp32-s3-devkitc-1
board_build.arduino.memory_type = qio_opi
board_upload.flash_size = 16MB
board_build.arduino.partitions = app3M_fat9M_16MB.csv

lib_deps = 
    https://github.com/rzeldent/esp32-smartdisplay.git#feature/esp32s3	

build_flags =
    -Ofast
    -Wall
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    -D LV_CONF_PATH=${PROJECT_INCLUDE_DIR}/lv_conf_new.h
    -D ESP_LCD_ST7796_VER_MAJOR=1
    -D ESP_LCD_ST7796_VER_MINOR=2
    -D ESP_LCD_ST7796_VER_PATCH=0
    -D ESP_LCD_ILI9341_VER_MAJOR=1
    -D ESP_LCD_ILI9341_VER_MINOR=2
    -D ESP_LCD_ILI9341_VER_PATCH=0
    -D ESP_LCD_GC9A01_VER_MAJOR=1
    -D ESP_LCD_GC9A01_VER_MINOR=2
    -D ESP_LCD_GC9A01_VER_PATCH=0
    -D CONFIG_ESP_LCD_TOUCH_MAX_POINTS=1
    -D CONFIG_XPT2046_CONVERT_ADC_TO_COORDS
    -D CONFIG_XPT2046_Z_THRESHOLD=600
    -D ESP32_8048S070C
    -D BOARD_HAS_PSRAM     
```

my code 
```c++
#include "esp_log.h"
#include <esp32_smartdisplay.h>
void setup()
{
    Serial.begin(115200);
    delay(1000);
    ESP_LOGI("Main", "smartdisplay_init");
    smartdisplay_init();
    ESP_LOGI("Main", "ui_init");
}

void loop()
{}
```

after compiling and upload i got next error with repeated reboot

E (1007) lcd_panel.rgb: esp_lcd_new_rgb_panel(151): no mem for frame buffer

```stacktrace
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420611fe
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
[   152][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  1157][I][main.cpp:27] setup(): [Main] smartdisplay_init
E (1007) lcd_panel.rgb: esp_lcd_new_rgb_panel(151): no mem for frame buffer
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x4037b490
file: ".pio/libdeps/esp32-s3-devkitc-1/esp32_smartdisplay/src/smartdisplay_rgb.c" line 29
func: lvgl_tft_init
expression: esp_lcd_new_rgb_panel(&tft_panel_config, &panel_handle)

abort() was called at PC 0x4037b493 on core 1

Backtrace: 0x40377dc6:0x3fceba50 0x4037b49d:0x3fceba70 0x403813d5:0x3fceba90 0x4037b493:0x3fcebb10 0x420276c1:0x3fcebb30 0x42027433:0x3fcebbf0 0x42002431:0x3fcebc10 0x4202a84e:0x3fcebc50

```

do you have any ideas how to deal with this problem?

rzeldent, 2023-12-06:
Hi, I checked the config, see line below for the ESP32_8048S070:
#define RBG_PANEL_CONFIG {.clk_src=LCD_CLK_SRC_PLL160M,.timings={.pclk_hz=12000000,.h_res=TFT_WIDTH,.v_res=TFT_HEIGHT,.hsync_pulse_width=30,.hsync_back_porch=16,.hsync_front_porch=210,.vsync_pulse_width=13,.vsync_back_porch=10,.vsync_front_porch=22,.flags={.hsync_idle_low=1,.vsync_idle_low=1,.pclk_active_neg=1,}},.data_width=16,.sram_trans_align=8,.hsync_gpio_num=39,.vsync_gpio_num=40,.de_gpio_num=41,.pclk_gpio_num=42,.data_gpio_nums={15, 7, 6, 5, 4, 9, 46, 3, 8, 16, 1, 14, 21, 47, 48, 45},.disp_gpio_num=-1}

The unsigned int fb_in_psram: 1 means it is only one bit (a flag) . This is the definition of the structure, no constructor.

Anyway, the flag is not set, so by default 0, (false). So this looks okay.

Each pixel requires 2 bytes, so 800 pixels * 16 lines * 2 = 25Kb frame buffer.
I think the frame buffer is too big. Try setting the LVGL_PIXEL_BUFFER_LINES from 16 to 8, this might help... This depends also on the size of your program. This is located in the file smartdisplay.h.

// LVGL lines buffered
#define LVGL_PIXEL_BUFFER_LINES 16

Let me know if this works!

sag3ll0, 2023-12-07:
Hello, is not work, no changes

[image]

sag3ll0, 2023-12-07:
After changing the above, compiling and uploading to the board, the screen starts blinking unevenly, and turns white after a couple of minutes

sag3ll0, 2023-12-07:
i found that freezed in lvgl_tft_init function
[image]

rzeldent, 2023-12-07:
Hi Oleg,

Also here, the code is out of date please update. Try to delete the .pio directory.
Then it should be downloaded again and hopefully refreshed!

sag3ll0, 2023-12-08:
> Maybe you can try it with the feature branch of the demo application?

same error

[image]

Luc3as, 2023-12-09:
Thank you gentlemen, 
I went down the same path and ran into the same errors, your suggestions and fixes allowed me to successfully run the DEMO from Squareline studio on the display. 
thank you and have a nice day :beers:

Luc3as, 2023-12-10:
Something went wrong , using latest version of demo and library feature s3 branch , model of display is ESP32_8048S070C , 
Yesterday I tried the demo and it was working correctly when I added fb psram flag to smartdisplay_rgb.c
Now I have this file from repo without the flag 

if I can enable debug to see what could cause issue please advise how

https://github.com/rzeldent/esp32-smartdisplay/assets/6381468/cbafacea-ea5d-4119-bd93-e1050330efba

rzeldent, 2023-12-10:
Hi Luc3as,

I think it is the timing of the pclk. I updated this to 32Mhz (trhis worked on the 4827S043) but looks like this is a problem on the ESP32_8048S070C. So the timing has been restore to: timings={.pclk_hz=16000000. Additionally the psram flag has been set.

Let me know if this resolves the issue!

Luc3as, 2023-12-10:
yes, looks crisply sharp now , thank you :) btw, is there way to rotate the
screen to vertical orientation?

rzeldent, 2023-12-10:
Hi Luc3cas,

I checked the driver and the stock driver does not support rotation. It is a pity because all the other drivers support this!
I think it is possible to do this in the software by changing the way how things are drawn...

Let me think about this...

SiWoC, 2023-12-10:
I see you guys are hard at work!
Thumbs up.
Can't wait to see this committed/available, tried a fresh feature/esp32s3 but got the same errors.

Keep up the good work.

rzeldent, 2023-12-10:
Well, software rotation seems to be in LVGL. Made some changes to enable this.
Works on my ESP32-4827S043. Give it a try!

sag3ll0, 2023-12-11:
Hi @rzeldent, 
about latest demo on my `ESP32_8048S070C` if i build and upload demo with config

[image]

i got error

[image]

for fix it i need add config (and demo work good)
```ini
[env:esp32-s3-devkitc-1]
board = esp32-s3-devkitc-1

board_build.arduino.memory_type = qio_opi

build_flags =
    ${env.build_flags}
    -D BOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    # Smartdisplay selection
    -D ESP32_8048S070C

lib_deps = ${env.lib_deps}
```

sag3ll0, 2023-12-11:
My configuration that I usually use for the board ESP32_8048S070C  looks like this

```ini
[env:esp32-s3-devkitc-1]
board = esp32-s3-devkitc-1

board_build.arduino.memory_type = qio_opi
board_upload.flash_size = 16MB
board_build.arduino.partitions = app3M_fat9M_16MB.csv
board_build.filesystem = littlefs

build_flags =
    ${env.build_flags}
    -D BOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    # Smartdisplay selection
    -D ESP32_8048S070C

lib_deps = ${env.lib_deps}
```

Maybe it will be useful to someone

SiWoC, 2023-12-11:
I can confirm that building with memory_type=qio_opi probably does the trick.
boards/esp32-s3-devkitc-1.json doesn't have that.

other minor difference between my board esp-sunton.json
[esp32-sunton.json](https://github.com/rzeldent/esp32-smartdisplay/files/13640777/esp32-sunton.json)
which I got from https://github.com/mhaberler/squareline-esp32/tree/main/boards

rzeldent, 2023-12-12:
You're right!

I created a new board definition: esp32-s3-devkitc-1-n16-psram.json. This has the correct amount of Flash and PSRAM.
No additional settings are required after INSTALLING the board.

Updated the README.md how to do this.

closed 2023-12-12 by rzeldent

sag3ll0, 2023-12-12:
maybe s3

[image]

it is also not necessary to store the description of the new boards in the platformio catalog, this can be done in the root of the project

[image]

sag3ll0, 2023-12-12:
for example my boards
[esp32_3248s035c.json](https://github.com/rzeldent/esp32-smartdisplay/files/13645175/esp32_3248s035c.json)
[esp32_8048s070c.json](https://github.com/rzeldent/esp32-smartdisplay/files/13645176/esp32_8048s070c.json)

rzeldent, 2023-12-12:
Hi Oleg,

Thanks for pointing out the mistake. Normally it is TLDR; 😉.

I will certainly update the board definition in the main project! Thanks!

Rene