rzeldent/esp32-smartdisplay #57: doesn't work with ESP32-2432S028R

ypyly, 2023-12-20:
Hi, I've tried the demo code and a regular one (generated by the SquareLine Studion) however the display never gets it's backlight on 
and there's a loop of the errors "indev_pointer_proc: X is 240 which is greater than hor. res    (in lv_indev.c line #362)" in both the example and the regular one.
in the demo code, the RGB LED on the back changes its colors.

rzeldent, 2023-12-21:
Hi ypyly,

I do not know exactly what you did. dev_pointer_proc: X is 240 means something is not right with the touch. No backlight is something wrong with the GPIO. Cannot determine what's wrong like this.
Do you have touch? Which version are you using, have you pulled the repo or using the PlatformIo registry? Is the board correct?

rzeldent, 2023-12-21:
Reproduced!

rzeldent, 2023-12-21:
Found issue, is a typo in the platformio file:
[image]

Must be 

```
[env:esp32-2432S028R]
board = esp32-2432S028R
```

Is fixed in main demo application. Explains why I did not have this issue as I do not use the demo all the time!

Enjoy

closed 2023-12-21 by rzeldent

ypyly, 2023-12-22:
there's a same typo in examples in this repo, I've tried to create a branch with a fix, but your settings forbid to push it :)
https://github.com/rzeldent/esp32-smartdisplay/blob/d224c97bde5efea8254de25d78076af801a6169e/examples/ci/platformio.ini#L89

rzeldent, 2023-12-22:
Hi ypyly,

Yes, I just fixed it but this is just for CI integration and is not used
elsewhere.
Demo application was already fixed.

GregoryHlavac, 2024-01-19:
Unsure what exactly this is happening for but getting a white screen with the demo now. (on v2 model but still).

rzeldent, 2024-01-19:
Hi Gregory,

There were also some changes in the defines. Maybe you can do a fresh clone
(with submodules)...

GregoryHlavac, 2024-01-20:
> Hi Gregory,
> 
> There were also some changes in the defines. Maybe you can do a fresh clone
> (with submodules)...
> 
> 
> 
> 
> > Unsure what exactly this is happening for but getting a white screen with
> > the demo now. (on v2 model but still).

Fresh as in from what date? Just tried this last night.

rzeldent, 2024-01-20:
Hi Gregory,

Just checked the normal ESP32-2432S028R. No problems.
Also compiled the v2 version and tried it on my board. Also seems to work fine.

Added a special log line to check if the special vendor config was present (see log) and this was the case.

```
[  2834][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
[  2842][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  2910][I][lvgl_ili9341.c:67] lvgl_lcd_init(): Vendor config: Present
[  3404][D][esp32-hal-adc.c:194] __analogReadMilliVolts(): eFuse Vref: Supported
[  3405][I][esp32-hal-adc.
![PXL_20240120_225314265](https://github.com/rzeldent/esp32-smartdisplay/assets/7620231/6ca86a76-37ec-43c8-bd13-d1bc84a2e70a)
c:235] __analogReadMilliVolts(): ADC1: Characterized using eFuse Vref: 1100
```
[image]

Maybe some problems including the boards submodules? That happens to me quite often...

Good luck

GregoryHlavac, 2024-01-21:
Fresh download on the board submodule, but upped the lv_conf log level to info; turned off wifi to remove that for now and re-ran demo and..
`--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[   280][I][main.cpp:26] setup(): CPU: ESP32-D0WD-V3 rev3, CPU Freq: 240 Mhz, 2 core(s)
[   281][I][main.cpp:27] setup(): Free heap: 311472 bytes
[   283][I][main.cpp:28] setup(): Free PSRAM: 0 bytes
[   287][I][main.cpp:29] setup(): SDK version: v4.4.6-dirty
[Info]  (0.293, +293)    lv_init: begin         (in lv_obj.c line #106)
[Info]  (0.788, +495)    lv_obj_create: begin   (in lv_obj.c line #215)
[Info]  (0.789, +1)      lv_obj_create: begin   (in lv_obj.c line #215)
[Info]  (0.789, +0)      lv_obj_create: begin   (in lv_obj.c line #215)
[Info]  (0.799, +10)     lv_obj_create: begin   (in lv_obj.c line #215)
[Info]  (0.801, +2)      lv_obj_create: begin   (in lv_obj.c line #215)
[Info]  (0.807, +6)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.814, +7)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.820, +6)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.826, +6)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.832, +6)      lv_btn_create: begin   (in lv_btn.c line #51)
[Info]  (0.837, +5)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.844, +7)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.850, +6)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (0.856, +6)      lv_img_create: begin   (in lv_img.c line #60)
[Info]  (0.861, +5)      lv_obj_create: begin   (in lv_obj.c line #215)
[Info]  (0.867, +6)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (0.883, +16)     lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (0.886, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (0.890, +4)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (0.898, +8)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (1.019, +121)    lv_label_create: begin         (in lv_label.c line #75)
[Info]  (1.024, +5)      lv_label_create: begin         (in lv_label.c line #75)
[Info]  (1.029, +5)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (1.033, +4)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (1.394, +361)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (1.397, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (1.714, +317)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (1.717, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (2.034, +317)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (2.354, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (2.357, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (2.674, +317)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (2.677, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (2.994, +317)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (3.314, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (3.634, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (3.637, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (3.954, +317)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (4.274, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (4.594, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (4.914, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (5.234, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (5.237, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (5.554, +317)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (5.874, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (5.877, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (6.194, +317)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (6.197, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (6.514, +317)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (6.834, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (7.154, +320)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (7.157, +3)      lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (7.474, +317)    lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[... cut ...]
[Info]  (12.918, +3)     lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)
[Info]  (13.235, +317)   lv_obj_update_layout: Layout update begin      (in lv_obj_pos.c line #314)`

Clearly **something** is happening because the lvgl stuff seems to be actively doing something but the screen still remains white; early steps into this ESP32 dev stuff best suggestion of where to probably look? (Your library looked so much more pleasant than dealing with all of the other low-level stuff)

rzeldent, 2024-01-21:
I see you're running SDK version: v4.4.6-dirty. Maybe this is something?

Try to update 6.5.0:
[image]

There are no messages errors, just lvgl update messages; no display or touch related errors...

AWoelfel, 2024-01-21:
I have exaclty the same problem as @GregoryHlavac with my ESP32-2432S028R.
I started from scratch (today); this is my first contact with any of the CYDs.
So i have pulled the most resent version for the demo as well as for the boards.

I can confirm : compilation and flashing works for both; v1 & v2. 
Both stay in the white screen after booting the display.

Checked the LV_conf logs as well! i see the activity as describe in the prev post.

GregoryHlavac, 2024-01-22:
Just tried to rebuild it and flash once more on a bare raspberry pi to rule out my desktop/windows nonsense; same problem.

I'm really not sure where that sdk-version 4.4.6-dirty is coming from but it seems to be part of the espressif 6.5.0 on pio.

rzeldent, 2024-01-22:
Hi,

The sdk-version 4.4.6-dirty is indeed part of espressif 6.5.0 and this is normal. I thought this might be something but is not.
Are you compiling for the right board?