rzeldent/esp32-smartdisplay #59: ESP32-2432S028 Board with USB-C not working

JoDaNl, 2023-12-26:
Got my ESP32-2432S028 board last week via Aliexpress. This board has, next o micro-usb also an USB-C connector.
Unfortunately I cannot get it working using this library. The screen light up...but remains white.

I ran a test with teh TFT_eSPI linbrary...and found oout that when selecting ILI9431_2_DRIVER I can use the display. Colors are inverted & screen is rotated but this can easily be corrected using:

  tft.invertDisplay(1);
  tft.setRotation(1);

Any thought on this ?

rzeldent, 2023-12-26:
Hi JoDaNI,

I looked it up: [ILI9341_Init.h - Strange graphic issues depending on TFT · Issue #1172 · Bodmer/TFT_eSPI (github.com)](https://github.com/Bodmer/TFT_eSPI/issues/1172)

Issue is caused by differences in chipsets. 

If you want you can change the initialization sequence in the file esp_lcd_ili9341.cpp. However, this is from the official EspressIf repo.
It is possible to supply an alternative init sequence. This cab be done by providing an initialization sequence in the file lvgl_ili9341.c calling  ESP_ERROR_CHECK(esp_lcd_new_panel_ili9341(io_handle, &panel_dev_config, &panel_handle));

If supply me with a working init sequence identical to the one defined in esp_lcd_ili9341.c, I can make a new alternative board definition.

typedef struct {
    int cmd;                /*<! The specific LCD command */
    const void *data;       /*<! Buffer that holds the command specific data */
    size_t data_bytes;      /*<! Size of `data` in memory, in bytes */
    unsigned int delay_ms;  /*<! Delay in milliseconds after this command */
} ili9341_lcd_init_cmd_t;

lazzaroevan, 2023-12-27:
I also have the same issue, this is a [link ](https://www.aliexpress.us/item/3256805429758720.html?spm=a2g0o.order_list.order_list_main.5.7fa418024Zu65T&gatewayAdapt=glo2usa#nav-specification)to the boards that I bought. How would I go about finding the driver for this?

JoDaNl, 2023-12-27:
Hi,

Have succesfully enabled the display of teh CYD-module with USB-C.
I used the following init-sequence:

```
static const ili9341_lcd_init_cmd_t vendor_specific_init_ILI9431_2[] = {
    {0xCF, (uint8_t []){0x00, 0xC1, 0X30}, 3, 0},
    {0xED, (uint8_t []){0x64, 0x03, 0X12, 0X81}, 4, 0},
    {0xE8, (uint8_t []){0x85, 0x00, 0x78}, 3, 0},
    {0xCB, (uint8_t []){0x39, 0x2C, 0x00, 0x34, 0x02}, 5, 0},
    {0xF7, (uint8_t []){0x20}, 1, 0},
    {0xEA, (uint8_t []){0x00, 0x00}, 2, 0},
    {0xC0, (uint8_t []){0x10}, 1, 0},
    {0xC1, (uint8_t []){0x00}, 1, 0},
    {0xC5, (uint8_t []){0x30, 0x30}, 2, 0},
    {0xC7, (uint8_t []){0xB7}, 1, 0},
    {0x3A, (uint8_t []){0x55}, 1, 0},
    {0x36, (uint8_t []){0x08}, 1, 0},
    {0xB1, (uint8_t []){0x00, 0x1a}, 2, 0},
    {0xB6, (uint8_t []){0x08, 0x82, 0x27}, 3, 0},
    {0xF2, (uint8_t []){0x00}, 1, 0},
    {0x26, (uint8_t []){0x01}, 1, 0},
    {0xE0, (uint8_t []){0x0F,0x2A,0x28,0x08,0x0E,0x08,0x54,0xA9,0x43,0x0A,0x0F,0x00,0x00,0x00,0x00}, 15, 0},
    {0xE1, (uint8_t []){0x00,0x15,0x17,0x07,0x11,0x06,0x2B,0x56,0x3C,0x05,0x10,0x0F,0x3F,0x3F,0x0F}, 15, 0},
    {0x2B, (uint8_t []){0x00, 0x00, 0x01, 0x3f}, 4, 0},
    {0x2A, (uint8_t []){0x00, 0x00, 0x00, 0xef}, 4, 0},
    {0x21, NULL, 0, 0},   // Inversion on
    {0x11, NULL, 0, 120},
    {0x29, NULL, 0, 1}
};
```

Next to this the display has it's contents by default rotated. To correct I defined the following in my platformio.ini build-flags:

```
    -DLCD_SWAP_XY=true 
    -DLCD_MIRROR_X=false
    -DLCD_MIRROR_Y=false
    -DLCD_WIDTH=320 
    -DLCD_HEIGHT=240
    -DTOUCH_DEV_CONFIG={.x_max=LCD_WIDTH,.y_max=LCD_HEIGHT,.rst_gpio_num=GPIO_NUM_NC,.int_gpio_num=GPIO_NUM_36}
    -DTOUCH_SWAP_X=true
    -DTOUCH_SWAP_Y=true
```

Now rotation is ok. Unfortunately this is a small offset in teh touch-screen coordinates. I will further investigate

lazzaroevan, 2023-12-27:
Thank you this worked, maybe we can run a calibration to fix it in software

rzeldent, 2023-12-28:
Hi JoDaNl,

I made a new board definition for the V2 version. See https://github.com/rzeldent/platformio-espressif32-sunton.
This board is also added to the develop branch of the demo. The custom initialization string is in the board definition and no code has to be updated (for the v2 version of the smartdisplay library).

The board name is ESP32-2432S028v2

I ran the code on my (normal/micro-usb) device and get the strange display that you had. So minus + minus is plus and think this will work for you.

Demo code for develop can be found at: https://github.com/rzeldent/esp32-smartdisplay/tree/develop

Hop this works, let me know.

JoDaNl, 2023-12-29:
I can confirm I get correct colors on the USB-C version. The image is in the same orientation as shown above.

I'm curious (as this is my first go with these type of boardxs & LVGL)...is this the correct orientation, or should the demo by default be in landscape format in stead of portrait?

rzeldent, 2023-12-30:
This orientation is the default 240x320, hence 2432S028 (cable on the bottom). You can change the orientation in the demo by enabling one of the lines:

```
    auto disp = lv_disp_get_default();
    // lv_disp_set_rotation(disp, LV_DISP_ROT_90);
    // lv_disp_set_rotation(disp, LV_DISP_ROT_180);
    // lv_disp_set_rotation(disp, LV_DISP_ROT_270);
```

JoDaNl, 2023-12-30:
Ok, thanks for your response!

So I tried the 'landscape' layout by  uncommenting  the line containing _lv_disp_set_rotation(disp, LV_DISP_ROT_90);_
This works as eepected.

Unfortunately there's a small ofsset in the touch screen, in both the X & Y direction. Does you library provide some means for correcting this ?

rzeldent, 2023-12-30:
I am in the process of adding 3 point calibration corrections. Code is present but still struggling with the GUI.
Take a look in feature/calibration

closed 2024-01-01 by rzeldent

rzeldent, 2024-04-16:
Yes,

There are two variables called; DISPLAY_GAP_X and DISPLAY_GAP_Y that can be used for an offset...