rzeldent/esp32-smartdisplay #60: Weird colors with esp32-8048S070C

Andre-Schuiki, 2023-12-26:
Hi all,

i tried the demo application but can't get the screen to work correct.
My colors looking weird...any ideas what i'm doing wrong?

[image]

This is my platformio.ini
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = esp32-8048S070C

[env]
platform = espressif32
framework = arduino

monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_filters = esp32_exception_decoder

# Partition scheme for OTA
board_build.partitions = min_spiffs.csv

build_flags =
    -Ofast
    -Wall
    -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    # LVGL settings
    -DLV_CONF_PATH=${PROJECT_INCLUDE_DIR}/lv_conf.h

#lib_deps = rzeldent/esp32_smartdisplay
lib_deps =
    https://github.com/rzeldent/esp32-smartdisplay.git
    https://github.com/schreibfaul1/ESP32-audioI2S

[env:esp32-8048S070C]
board = esp32-8048S070C
```

rzeldent, 2023-12-26:
Hi Andre,

Looks like the LV_COLOR16_SWAP is incorrect.

Andre-Schuiki, 2023-12-26:
Yeah that's what i thought too but i already tried with swap enabled and get the same result.

I will try it again tomorrow and give you feedback.

Thanks for your help!

rzeldent, 2023-12-26:
It can be that you need to rebuild the library. The define is only taken into account when compiling thew library (that is cached).
Retry it but delete the .pio directory first. It is also possible that you need to change the settings in Squareline to be able to compile...
This should have been solved in the 2.0.x versions?!

Andre-Schuiki, 2023-12-29:
> It can be that you need to rebuild the library. The define is only taken into account when compiling thew library (that is cached). Retry it but delete the .pio directory first. It is also possible [...]

Ok i have tried with LV_COLOR_16_SWAP 1 but i get the same result. (i also have deleted the .pio folder before compiling)

IanSC, 2024-01-07:
I have the same issue as this. Edited lv_conf.h and changed

`#define LV_COLOR_16_SWAP 0`

Did a PIO: Clean by clicking the trash can before uploading. Got the correct photo color,
but text are garbled and the colors for widgets are off.
eg. green border, greenish background, avocado button

### esp32-8048S070C

[image]

[image]

[image]
   
<br/><br/>
Tried it on another device and it appears correct (set the swap var to 1 since SPI).

### esp32-3248S035C

[image]

<br/><br/>
Not sure if related, but the downloaded library does not contain ST7701 files.

Thanks for the help and the great library!

rzeldent, 2024-01-07:
Hi IanSC,

Maybe the 16 bits color array is swapped, that's what I suspect. You can try to swap it by manually by setting in lvgl_rgb_panel.c the color 16 swap, just for testing:
```
// Create direct_io panel handle
#if LV_COLOR_16_SWAP != 0
```

Ordered the 8048S070C and will hopefully be able to reproduce it and correct it.

IanSC, 2024-01-07:
Hi rzeldent,

**A**: `lv_conf.h`   `#define LV_COLOR_16_SWAP <value>`
**B**: `lvgl_rgb_panel.c, line 28`   `#if LV_COLOR_16_SWAP <operator> 0`

|  A < value > | B < operator > | Image | Text | Colors |
|---|---|---|---|---|
| 1 | != | inverted | garbled | off |
| 0 | != | okay | garbled | off |
| 1 | == | inverted | garbled | off |
| 0 | == | okay | garbled | off |

Thanks for checking this out.

rzeldent, 2024-01-07:
Hi LanC,

Thanks for testing this. I was hoping it was just a typo... So basically the text is always garbled.... 
Have you tried inverting the image? esp_lcd_panel_invert_color(handle, true);

IanSC, 2024-01-08:
Hi rzeldent,

Took me a while to figure out where to the add the line of code.

I also tried the other cases from your previous suggestion.
I doubted there would be any difference since it's a text issue not color,
but surprisingly it got much better.

Thanks again!

in `lvgl_rgb_panel.c`, added:
`ESP_ERROR_CHECK(esp_lcd_panel_invert_color(panel_handle, true));`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;  at the end of the function `lvgl_lcd_init()`

**A**: `lv_conf.h`   `#define LV_COLOR_16_SWAP <value>`
**B**: `lvgl_rgb_panel.c, line 28`   `#if LV_COLOR_16_SWAP <operator> 0`

| CASE |  A < value > | B < operator > | Image | Text | Colors | Shade | Note |
|:---|:---|:---|:---|:---|:---|:---|:---|
| 1 | 1 | == | inverted | garbled | off | none | |
| 2 | 0 | == | wrong | garbled | off | purple | |
| 3 | 1 | != | wrong | okay | off | dark purple | |
| 4 | 0 | != | inverted | okay | okay | none | FPS overlay seems off |

### CASE 1
[image]

### CASE 2
[image]

### CASE 3
[image]

### CASE 4
[image]

IanSC, 2024-01-08:
Additional Test:
Just reporting here to help solve any potential issues.
I don't remember the colors being so far off with the built in app
that came with the device.

There are not bug reports, just observations.
Big screen is yellowish and less brighter.
Added:
```
void setup() {
    ...
    smartdisplay_init();
    // smartdisplay_lcd_set_backlight(1.0f); // screen flickers
    // smartdisplay_lcd_set_backlight(0.9f); // screen flickers
    smartdisplay_lcd_set_backlight(0.5f);    // screen flickers
    ...
    tutorialCode();
}
void loop() {
    lv_timer_handler();
}
```
Changing the brightness causes the screen to flicker and become unstable, even if set to 0.5f
which is the same inside smartdisplay_init(). I might be calling it in the wrong place.

Button test seems normal, except for overlay.
Solid RGB colors are off.
<br/>

[image]
[image]
[image]
[image]

IanSC, 2024-01-15:
Hi rzeldent,

Based on:
Esp32-8048S070 display, different version
https://github.com/rzeldent/esp32-smartdisplay/issues/90

I tried the settings for v1.3 and the text and colors are correct.

For noobs like me:
- make a copy of ``esp32-8048S070C.json`` inside ``board`` folder
- rename as, say ``esp32-8048S070C-v1.3.json``
- edit and apply suggested changes in Issue #90
- edit ``platformio.ini`` and use ``board = esp32-8048S070C-v1.3``
- rebuild

IanSC, 2024-01-24:
Tried LVGL examples with the changes and it works.

Thanks to @rzeldent !

So guys, let's support or buy this guy a beer, without his efforts these boards will be 
useless or extremely difficult to use. Looking at you Sunton.

rzeldent, 2024-01-26:
Thanks!

BTW in the develop the v1.3 board will be the default. Think the v1.1 was some typo.

MrOptifine, 2024-03-06:
Any idea on how to fix the screen flickers?

rzeldent, 2024-03-06:
Hi Optifine,

I think the PCLK needs tweaking....