rzeldent/esp32-smartdisplay #61: display resets with latest firmware

klayus, 2023-12-27:
Using the latest version of the library and the smartdisplay demo i get this error.
My display is esp32-3248S035R

PI_FAST_FLASH_BOOT)
flash read err, 988
ets_main.c 384 
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

rzeldent, 2023-12-27:
Hi Klayus,

This looks like a flashing or wrong device. Even at startup it should mention info. See demo app:

```
void setup()
{
    delay(250);
    Serial.begin(115200);
    Serial.setDebugOutput(true);

    log_i("CPU: %s rev%d, CPU Freq: %d Mhz, %d core(s)", ESP.getChipModel(), ESP.getChipRevision(), getCpuFrequencyMhz(), ESP.getChipCores());
    log_i("Free heap: %d bytes", ESP.getFreeHeap());
    log_i("Free PSRAM: %d bytes", ESP.getPsramSize());
    log_i("SDK version: %s", ESP.getSdkVersion());
```

klayus, 2023-12-28:
This is what it says now.

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[    18][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   296][I][main.cpp:21] setup(): CPU: ESP32-D0WD-V3 rev3, CPU Freq: 240 Mhz, 2 core(s)
[   297][I][main.cpp:22] setup(): Free heap: 293184 bytes
[   299][I][main.cpp:23] setup(): Free PSRAM: 0 bytes
[   304][I][main.cpp:24] setup(): SDK version: v4.4.4
E (521) spi: spi_bus_initialize(756): SPI bus already initialized.
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4008869b
file: ".pio/libdeps/esp32-3248S035R/esp32_smartdisplay/src/lvgl_xpt2046.c" line 35
func: lvgl_touch_init
expression: spi_bus_initialize(TOUCH_SPI_HOST, &spi_bus_config, SPI_DMA_CH_AUTO)
E (563) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (564) lcd_panel.io.spi: panel_io_spi_tx_color(380): spi transmit (queue) color failed
E (567) st7796: panel_st7796_draw_bitmap(280): send command failed
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400886b0
file: ".pio/libdeps/esp32-3248S035R/esp32_smartdisplay/src/lvgl_st7796.c" line 29
func: st7796_lv_flush
expression: esp_lcd_panel_draw_bitmap(panel_handle, area->x1, area->y1, area->x2 + 1, area->y2 + 1, color_map)

abort() was called at PC 0x400886b3 on core 1

Backtrace: 0x40083a71:0x3ffb1fe0 0x400886bd:0x3ffb2000 0x4008e351:0x3ffb2020 0x400886b3:0x3ffb20a0 0x400f7c9a:0x3ffb20c0 0x4011cd39:0x3ffb20e0 0x400da85d:0x3ffb2110 0x400db221:0x3ffb21e0 0x400e73c0:0x3ffb2240 0x400d2893:0x3ffb2270 0x400fa405:0x3ffb2290

  #0  0x40083a71:0x3ffb1fe0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x400886bd:0x3ffb2000 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x4008e351:0x3ffb2020 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x400886b3:0x3ffb20a0 in _esp_error_check_failed at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_err.c:43
  #4  0x400f7c9a:0x3ffb20c0 in st7796_lv_flush at .pio/libdeps/esp32-3248S035R/esp32_smartdisplay/src/lvgl_st7796.c:29 (discriminator 1)
  #5  0x4011cd39:0x3ffb20e0 in call_flush_cb at .pio/libdeps/esp32-3248S035R/lvgl/src/core/lv_refr.c:1241
  #6  0x400da85d:0x3ffb2110 in draw_buf_flush at .pio/libdeps/esp32-3248S035R/lvgl/src/core/lv_refr.c:1216
      (inlined by) refr_area_part at .pio/libdeps/esp32-3248S035R/lvgl/src/core/lv_refr.c:712
  #7  0x400db221:0x3ffb21e0 in refr_area at .pio/libdeps/esp32-3248S035R/lvgl/src/core/lv_refr.c:594
      (inlined by) refr_invalid_areas at .pio/libdeps/esp32-3248S035R/lvgl/src/core/lv_refr.c:532
      (inlined by) _lv_disp_refr_timer at .pio/libdeps/esp32-3248S035R/lvgl/src/core/lv_refr.c:324
  #8  0x400e73c0:0x3ffb2240 in lv_timer_exec at .pio/libdeps/esp32-3248S035R/lvgl/src/misc/lv_timer.c:313
      (inlined by) lv_timer_handler at .pio/libdeps/esp32-3248S035R/lvgl/src/misc/lv_timer.c:109
  #9  0x400d2893:0x3ffb2270 in loop() at src/main.cpp:59
  #10 0x400fa405:0x3ffb2290 in loopTask(void*) at C:/Users/Dragos/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

rzeldent, 2023-12-28:
Yes, you're using the main repository of the boards, not the develop.
Anyway, already merged it to main so updating the boards should resolve it

klayus, 2023-12-28:
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[    17][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
E (257) spi: spi_bus_initialize(756): SPI bus already initialized.
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40087b23
file: ".pio/libdeps/esp32-3248S035R/esp32_smartdisplay/src/lvgl_xpt2046.c" line 35
func: lvgl_touch_init
expression: spi_bus_initialize(TOUCH_SPI_HOST, &spi_bus_config, SPI_DMA_CH_AUTO)
E (281) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (284) lcd_panel.io.spi: panel_io_spi_tx_color(380): spi transmit (queue) color failed
E (292) st7796: panel_st7796_draw_bitmap(280): send command failed
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40087b38
file: ".pio/libdeps/esp32-3248S035R/esp32_smartdisplay/src/lvgl_st7796.c" line 29
func: st7796_lv_flush
expression: esp_lcd_panel_draw_bitmap(panel_handle, area->x1, area->y1, area->x2 + 1, area->y2 + 1, color_map)

abort() was called at PC 0x40087b3b on core 1

Backtrace: 0x40083999:0x3ffb1fe0 0x40087b45:0x3ffb2000 0x4008d7b1:0x3ffb2020 0x40087b3b:0x3ffb20a0 0x400f6ef5:0x3ffb20c0 0x40118835:0x3ffb20e0 0x400d9d49:0x3ffb2110 0x400da70d:0x3ffb21e0 0x400e6764:0x3ffb2240 0x400d243f:0x3ffb2270 0x400f7e2d:0x3ffb2290

ELF file SHA256: e2fb990cb3c4746f

i'm guessing the same thing happens?

rzeldent, 2023-12-28:
Please do a pull of the boards git submodule, looks like this was not done.
Also, maybe restarting PlatformIo might help as the boards are loaded once when starting up.

klayus, 2023-12-28:
i deleted all the boards and downloaded them again, copied them in boards folder in .platformio\platforms\espressif32\boards
still the same issue. i'm not sure how git works but there not seem to be any changes from today, last change was 2 weeks ago. i tried downloading the boards from the develop repository and still them same error. the display works fine with 1.0.5 version of the library

rzeldent, 2023-12-28:
Hi klayus,

Try to download the develop version: https://github.com/rzeldent/esp32-smartdisplay-demo/tree/develop

or, in a command window in the demo directory type:
```
git checkout develop
```

In the boards directory there should be a esp32-3248S035Rv2 and in this file, the LCD_PANEL_DEV_CONFIG should finish with .init_cmds_size=23

klayus, 2023-12-29:
the file with the v2 at the end is esp32-2432S028Rv2, my screen is esp32-3248S035R, can i use the same configuration?

rzeldent, 2023-12-29:
No, that's for a board with usb connector and a different lcd display.

klayus, 2023-12-29:
let me rephrase, there is no  esp32-3248S035Rv2 json file in boards.

rzeldent, 2023-12-29:
Yes there is: https://github.com/rzeldent/platformio-espressif32-sunton
Maybe you need to add it in the ini?

rzeldent, 2023-12-29:
But this is for something completely different?!

klayus, 2024-01-09:
i tried it on a different computer and the demo does not have any issues. sorry, there seem to be something wrong with my vscode or platformio instalation. 
keep up the great work!!!

closed 2024-01-09 by klayus