rzeldent/esp32-smartdisplay #62: audio volume

sartyx, 2023-12-27:
Not really an issue but is it possible to include a volume control? Afaik the tone function got no parameters for the volume?

rzeldent, 2023-12-27:
There is no volume control as it is only one GPIO: on or off. What you can do is use the I2S and use the setVolume function of the audio library; this uses a DAC

sartyx, 2023-12-28:
Hmm I2S? Wouldn't that require more of the few GPIO pins?
I am already using two of them ;-)
I guess it is easier to build a little circuit with a pot, as I just want the overall level to be adjustable.

rzeldent, 2023-12-28:
No,

Take a look at the demo. It makes a webradio by using the internal DAC. This PWM is passed trough a low-pass filter and voila!

sartyx, 2023-12-28:
Thanks, this is confusing me now, I gotta check out the functionality auf audio.h

rzeldent, 2023-12-28:
Hi Sartyx,

You're right, was a renaming issue in the defines... Hang on...

rzeldent, 2023-12-28:
Can you update the boards definitions: goto the boards directory and do a git pull.
Then restart platformio

sartyx, 2023-12-28:
```
\boards>git pull
remote: Enumerating objects: 34, done.
remote: Counting objects: 100% (34/34), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 23 (delta 16), reused 21 (delta 15), pack-reused 0
Unpacking objects: 100% (23/23), 7.01 KiB | 163.00 KiB/s, done.
From https://github.com/rzeldent/platformio-espressif32-sunton
   67820f4..1109442  develop    -> origin/develop
   48c8c70..04b478b  main       -> origin/main
You are not currently on a branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>
```

Hmm, that didn't change much I guess. Still the same result.

rzeldent, 2023-12-28:
The boards directory is still on main.
Pushed the changes also to main

sartyx, 2023-12-28:
Hmm all branches now state they are not from today? 

Changing only to       "-DBOARD_HAS_SPEAK", doesn't solve the problem, so what else is changed?

rzeldent, 2023-12-28:
Nothing else. You can verify it in the git repo.

rzeldent, 2024-02-02:
Hi Sartyx,

Is it working now?

sartyx, 2024-02-03:
Haven't tested this any deeper, switched to another audio lib, however
I noticed another thing but that isn't really display related. I just came across that because the display code makes usage of the latest ESP32 libs. If you try to do a WIFI UDP communication between two of the displays it doesn't work but it does with the older libs. But that also happens using other ESP32, so it is a bug in the WIFI libs I guess.

closed 2024-02-10 by rzeldent