rzeldent/esp32-smartdisplay #70: Empty screen at ESP32-2432S032C

Keidran, 2023-12-31:
LED blinking
The screen backlights work, but the screen is empty (demo)

[   297][I][main.cpp:26] setup(): CPU: ESP32-D0WD-V3 rev3, CPU Freq: 240 Mhz, 2 core(s)
[   298][I][main.cpp:27] setup(): Free heap: 248652 bytes
[   300][I][main.cpp:28] setup(): Free PSRAM: 0 bytes
[   305][I][main.cpp:29] setup(): SDK version: v4.4.6-dirty

[image]
[image]

rzeldent, 2023-12-31:
Hi Keidran,

See this an cyd with USB-C. Do not have this one in possession. and you're using SDK 4.4.6 (I'm using SDK version: v4.4.5).
What application and branch are you using?

Keidran, 2023-12-31:
I build Demo from master.
Driver: ST7789 (from specifications)
Platform: Espressif 32 6.5.0
The screen backlights reacts to the change in light from the photoresistor.
Switching LV_COLOR_16_SWAP does not change anything.

rzeldent, 2023-12-31:
Will have to take a look. Are you using PlatformIo?

Keidran, 2023-12-31:
Yes

Keidran, 2024-01-02:
Okey. 
In ESP32-2432S032C.json, the ST7796 driver was set, its change to ST7789 solved the problem.
Also, I changed SPI_MODE0 to SPI_MODE3

rzeldent, 2024-01-02:
Yes, I checked the Sunton demo and the driver is indeed ST7796. Fixed in develop and just merged back to main.
If you update the submodule it should work.

I did not change the SPI_MODE3, this because there is also a  ESP32-2432S032R that uses the same bus...
Can do a check with another board to see this works also with the XPT2046.

I ordered one of those because is lacking in the collection of boards!

Thanks!

rzeldent, 2024-01-02:
Not all has been changed, just a sec...

rzeldent, 2024-01-02:
Also the other defines were changed. Tried the driver on the  ESP32-1732S019N. Works.

Also, on the ESP32-1732S019N tested the difference between SPI_MODE0 and SPI_MODE3. The timing for the data should be differences but I do not see any changing to SPI_MODE3. Is this really required?

Keidran, 2024-01-03:
Mb I have bad cyd but MODE0 and MODE1 don't work, MODE2 and MODE3 works

rzeldent, 2024-01-03:
Hi Keidran,

Thanks for the reply, I changed it in the driver to mode 3.
Maybe you can check if it works not as it should.

Keidran, 2024-01-03:
What exactly needs to be checked, modes?

rzeldent, 2024-01-03:
Well, updated the mode in the smartdisplay for the ST7789 to mode 3. Just a check to see if it works now on your device...

Keidran, 2024-01-03:
As I wrote above, it works, like mode 2
And turned off the lcd_mirror_x. Although this is already possible because of the board itself.

rzeldent, 2024-01-03:
Yes, that's also logical that the mirror_x is no longer required. Then also the touch mirror_x should be removed.
I'll also update this setting!

Keidran, 2024-01-03:
For the touch, it was not set

rzeldent, 2024-01-04:
Should also be fixed now in the definitions. Thanks for mentioning! 👍

closed 2024-01-07 by Keidran