rzeldent/esp32-smartdisplay #72: screen noise / weird screen behaviour in newer demo version

sartyx, 2024-01-01:
I switched to the newer demo today, because my code based on the old version and I got into more and more trouble with the dependencies.

I compiled the demo as it is and it works, then I changed the UI to my version and got some "static noise" on the screen. 
[image]

First I thought the modules I attached to the GPIO are responsible but then I switched back to the demo UI but included the libs for my attached modules. Still working. So did some changes to the original demo UI within squareline studio. Everything went ok until I put a PNG bitmap in background. The static noise came back.

Since the old demo version (where the rotation of the screen was still set in platformio.ini) something serious must have changed.
You mentioned new drivers for the display? 

I could imagine some memory issues that would cause this, my UI hat a bunch of dropdowns select boxes on the first screen.
And it was 16 bit not 16 bit SWAP.

The failures are different in type, on my UI it was mainly in the lower left part of the first screen, on full screen at the second screen. Adding a bitmap to the default UI it is at about 56 pixel from top of the screen.

[image]

rzeldent, 2024-01-01:
Hi sartyx,

Yes, there has been a switch to the new ESP lcd interface now present in the firmware. Now som heavy refactoring is done.
You are working on a esp32-2432S024C right?!

rzeldent, 2024-01-01:
Retested the esp32-2432S024C no problems observed. What is the device you're using?

sartyx, 2024-01-02:
Windows 10 / platformio / esp32-2432S024C
I'd get back to the older version but sadly I have overwritten everything and of course it ain't the best solution.

sartyx, 2024-01-02:
I just flashed another spare  esp32-2432S024C and the problem doesn't appear there. So I removed all the attached stuff (battery,speaker/sensor) from the other but the problem remains, I reflashed it and the problem remains.

Additionaly the malfunctioning display does a backlight fade after a few seconds, the new one doesn't. Using the same code.
There gotta be some remains of older flashs on the older display, don't know how to explain it otherwise.

rzeldent, 2024-01-02:
Are you maybe using the Squareline code?
Strange that if you have two  esp32-2432S024C's only one is having the problem, right?
Are there any differences usb-C/micro USB or different display?

There was  a similar issue with the esp32-2432S028 See: https://github.com/rzeldent/esp32-smartdisplay/issues/59 However, here there was an image...

sartyx, 2024-01-02:
No actually both having the problem. Just one more than the other. But as edited above, I just flashed the demo on both of them and both work. (They are visually the same, ordered at the same time.) 

However flashing them back to the squarline stuff now still causes static problems on the old version.
Note that I didn't have these issues when I used your old demo as "template" and pulled my squareline layout over it. 

Either the flashes in between messed it up or something important in the driver or lib changed.

What are your settings on the Squareline Studio project property page?

[image]

rzeldent, 2024-01-02:
I do not use the drivers from Squareline, just the UI generator:

[image]

sartyx, 2024-01-02:
I guess it doesn't really matter. No matter what I use there, using images causes trouble and the old display stays in trouble, no matter what, except when using the sunton demo.

I'll use the new display for now and keep the images away as I don't really need them for my purpose, would have been nice though. The backlight fading is obviously caused by battery-powering it. It just came back when I connected the battery.

rzeldent, 2024-01-02:
Might also be the new feature to have the backlight adapt to ambient light...
See manual

sartyx, 2024-01-02:
It is pretty annoying.

```
#define TFT_BL 21
#define TFT_BACKLIGHT_ON HIGH 
```
this is actually not doing anything

nor is this within setup():

smartdisplay_lcd_set_backlight(4000); or a percentage value 
smartdisplay_lcd_set_backlight(0.8); 

it simply fades to almost black, instantly when flashed

Don't exactly understand what you wrote on the introduction, it is confusing, why timer and then brightness?
I'd expect something like  smartdisplay_lcd_set_backlight(fadeinmilliseconds,topercentage) 

What manual? The one that comes with the display?

rzeldent, 2024-01-02:
https://github.com/rzeldent/esp32-smartdisplay?tab=readme-ov-file#void-smartdisplay_lcd_set_brightness_cbsmartdisplay_lcd_adaptive_brightness_cb_t-cb-uint-interval

sartyx, 2024-01-02:
Thanks, that keeps it at least at a level. I am too dumb to understand the purpose of the definition and the backlight function then. They are obsolete in my eyes, at least for THIS display.
The cbs is probably toned down in my case because I put the display in a box, so the sensor assumes total darkness.

sartyx, 2024-01-07:
This is getting wild, I just added 16 containers to a screen and the static noise came back. No graphics at all.
Removed the containers down to zero... the noise got less and less but is still there at a minimum amount.

Can't believe noone else runs into that trouble. 

I don't think my ui is very complex, there are just two screens now, first with 5 dropdowns and a button, second with just just 2 labels and a textarea.

So what did I do? I grabbed your squareline project and added 5 dropdowns, changed the background, added a second screen with some labels. 

No static noise. 
I added a background image to both screens. (2 times the same image). 
Static noise is back!
I remove the background image again and replaced it with ui_img_lenna_165_png. 
No static noise.
I made a new background in purple instead of blue with the same program (affinity photo) and added it.
No static noise.
I added 16 containers on the second screen.
No static noise.
I resaved the troublemaking background and added it as background. 
Static noise is back.
I changed the main colors from blue to red in the image.
Static noise is GONE!

**I have NO explanation for this but obviously the display or display driver got issues with blue.**
[trouble_ui.zip](https://github.com/rzeldent/esp32-smartdisplay/files/13854885/trouble_ui.zip)

I attached the modded ui with the trouble background, if you like to test this. Should work with your demo as I didn't change any of  the used objects.

rzeldent, 2024-01-07:
Hi Sartyx,

I was curious what was happening so tried your UI files to see what was going on... This is what I see:

[image]

It is not completely identical as this is a  esp32-2432S024**R** but the display should be...
I'm also using the latest version from github (develop).

Can you give me the code that creates trouble, then I can take a look!

sartyx, 2024-01-08:
Okay, I see you don't see a flickering white line on the left edge. The blue below right is a button switching to screen2, check if the line exists there. I don't have the frag/cpu stats running and I use the capacitive touch version.  Code doesn't seem to matter, it already appears if I reduce the code to this:

```

#include <Arduino.h>

#include <esp32_smartdisplay.h>
#include <ui/ui.h>
#include <Audio.h>

 

void OnButtonClicked(lv_event_t *e)
{
    static uint8_t cnt = 0;
    cnt++;
 
}

void setup()
{
    delay(250);
    Serial.begin(115200);

    Serial.println("launching...");

    Serial.setDebugOutput(true);
    smartdisplay_init();
    auto disp = lv_disp_get_default();
    lv_disp_set_rotation(disp, LV_DISP_ROT_90);
    smartdisplay_lcd_set_brightness_cb(NULL,10000);
    ui_init();
   
}

void loop()
{
  
lv_timer_handler();

}

```

rzeldent, 2024-01-08:
[image]

sartyx, 2024-01-08:
Looks like I got some "monday" editions of the display then. :(

sartyx, 2024-02-03:
No I guess the displays I got are just faulty about that, as long as noone else got that happening.

rzeldent, 2024-02-03:
What you can try is a lower ST7789_IO_I80_CONFIG_PCLK_HZ.

closed 2024-02-10 by rzeldent