rzeldent/esp32-smartdisplay #75: Issue with new clone of esp32_smartdisplay and lvgl_ili9341.c

lazzaroevan, 2024-01-02:
I was compiling the demo fine until all of a sudden this error popped up, and now it wont compile and none of my previous projects are compiling either.

`.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/lvgl_ili9341.c: In function 'lvgl_lcd_init':
.pio/libdeps/esp32-2432S028Rv2/esp32_smartdisplay/src/lvgl_ili9341.c:49:180: error: expected ',' or ';' before '}' token
     const esp_lcd_panel_dev_config_t panel_dev_config = {.reset_gpio_num = -1, .color_space = ESP_LCD_COLOR_SPACE_BGR, .bits_per_pixel = 16, .vendor_config = ILI9341_VENDOR_CONFIG};
                                                                                                                                                                                    ^
Compiling .pio\build\esp32-2432S028Rv2\lib63e\FFat\FFat.cpp.o
*** [.pio\build\esp32-2432S028Rv2\libb84\esp32_smartdisplay\lvgl_ili9341.c.o] Error 1`

Do you have any Idea what happened?

rzeldent, 2024-01-02:
The board definitions contained structures with the definition. Not all compilers could handle this (mac could not compile)
These structures are now initialized in the code but the pins and timings in the boards definition.

To fix this update the smartdisplay library and the boards submodule to the latest main.

Hope this helps.

lazzaroevan, 2024-01-02:
|-- esp32_smartdisplay @ 2.0.2+sha.cb0cf33
Even with a newly built fresh clone the demo is still not working and throwing the same error, do you know what could be causing this? I am using the new esp32_2432S028v2 board that you just updated for.

rzeldent, 2024-01-02:
Yes, something i'm workin on. The develop branch of the boards has this fixed already

lazzaroevan, 2024-01-02:
Gotcha, Thank you

rzeldent, 2024-01-02:
Fix also merged to main. Was still looking for another way to fix this. This does not work well in CI builds...

rzeldent, 2024-02-02:
Fixed with single quotes in defines

closed 2024-02-02 by rzeldent