rzeldent/esp32-smartdisplay #90: Esp32-8048S070 display, different version

Reneg973, 2024-01-14:
Hi,
there is a different version of this display, see https://www.makerfabs.com/sunton-esp32-s3-7-inch-tn-display-with-touch.html.

The colors were not right, but the image appeared to be ok, so I just changed the json color pins to this:
      "-DRGB_PANEL_R0=14",
      "-DRGB_PANEL_R1=21",
      "-DRGB_PANEL_R2=47",
      "-DRGB_PANEL_R3=48",
      "-DRGB_PANEL_R4=45",
      "-DRGB_PANEL_G0=9",
      "-DRGB_PANEL_G1=46",
      "-DRGB_PANEL_G2=3",
      "-DRGB_PANEL_G3=8",
      "-DRGB_PANEL_G4=16",
      "-DRGB_PANEL_G5=1",
      "-DRGB_PANEL_B0=15",
      "-DRGB_PANEL_B1=7",
      "-DRGB_PANEL_B2=6",
      "-DRGB_PANEL_B3=5",
      "-DRGB_PANEL_B4=4",

Touch seems to work, at least the button on your demo.
Could you provide a picture of how the screen looks correctly?

Reneg973, 2024-01-14:
Note, on the wiki page they write about version 1.0 and 1.1. I own a version 1.3, but it seems compatible to version 1.1

rzeldent, 2024-01-17:
Hi Rene,

Thanks for noticing this! Mystery solved!

I split (in the develop branch) the 8048S070C into two versions v11 and v13.

After this has been merged in the main branch, then so you'll need to update the board you're using in the platformio.ini file.

rzeldent, 2024-02-07:
Think this was a typo. Now back to one version.

closed 2024-02-07 by rzeldent