rzeldent/esp32-smartdisplay #91: touch area not fullscreen anymore ~320x200 instead of 320x240

sartyx, 2024-01-15:
With the esp32-2432S024C the toucharea is messed up now. (Updated the lib on last wekend with the new board definitions.)

In panoramic orientation about 1/5 of the lower edge doesn't response to touches. feels like somewhere is a 320x200 rather than 320x240 set.

I thought my display f****d up further but all of the 3 displays I got show that behaviour.

sartyx, 2024-01-15:
- reinstalled platformio
- reexported from squarline studio
- reduced the sqs UI to one screen
 No change.
- Tried the demo from the display, works with full screen touch. 

Something is messed up with the touch!

rzeldent, 2024-01-17:
HI Sartyx,

Good job!

I checked the code the code and ordered a 2432S024C to test it myself. Not arrived yet :-(
What you say is very well possible. But before I update the repository, can you also check if
```
      "-DLCD_MIRROR_X=true",
      "-DLCD_MIRROR_Y=false",
```
are still correct?

This because it is strange that only the touch is flipped with respect to the display. Is possible but would be the first.
Can you provide me with the complete defines?

sartyx, 2024-01-18:
check your older version it was like this before ;-) 
BUT, that might only be a charge of their displays, who knows... maybe add a comment somewhere in your docs ?

```
{
  "build": {
    "arduino": {
      "ldscript": "esp32_out.ld"
    },
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_ESP32_DEV",
      "-DESP32_2432S024C",
      "-DLCD_WIDTH=240",
      "-DLCD_HEIGHT=320",
      "-DBCKL=27",
      "-DLCD_USES_ILI9341",
      "-DILI9341_SPI_HOST=SPI2_HOST",
      "-DILI9341_SPI_MOSI=13",
      "-DILI9341_SPI_MISO=12",
      "-DILI9341_SPI_SCLK=14",
      "-DILI9341_CS=15",
      "-DILI9341_DC=2",
      "-DILI9341_VENDOR_CONFIG=NULL",
      "-DLCD_SWAP_XY=false",
      "-DLCD_MIRROR_X=true",
      "-DLCD_MIRROR_Y=false",
      "-DBOARD_HAS_TOUCH",
      "-DTOUCH_USES_CST816S",
      "-DCST816S_I2C_HOST=0",
      "-DCST816S_I2C_SDA=33",
      "-DCST816S_I2C_SCL=32",
      "-DCST816S_RST=25",
      "-DCST816S_INT=21",
      "-DTOUCH_SWAP_X=false",
      "-DTOUCH_SWAP_Y=false",
      "-DBUTTON_BOOT=0",
      "-DBOARD_HAS_TF",
      "-DTF_CS=5",
      "-DTF_SPI_MOSI=23",
      "-DTF_SPI_SCLK=18",
      "-DTF_SPI_MISO=19",
      "-DBOARD_HAS_RGB_LED",
      "-DRGB_LED_R=4",
      "-DRGB_LED_G=16",
      "-DRGB_LED_B=17",
      "-DBOARD_HAS_CDS",
      "-DCDS=34",
      "-DBOARD_HAS_SPEAK",
      "-DSPEAK=26"
    ],
    "f_cpu": "240000000L",
    "f_flash": "40000000L",
    "flash_mode": "dio",
    "mcu": "esp32",
    "variant": "esp32"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "debug": {
    "openocd_board": "esp-wroom-32.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "esp32-2432S024C",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.aliexpress.com/item/1005005865107357.html",
  "vendor": "Sunton"
}
```

rzeldent, 2024-02-02:
Fixed

closed 2024-02-02 by rzeldent