rzeldent/esp32-smartdisplay #104: Compilation issue with touch & ESP32-8048S070

dronecz, 2024-01-21:
Hello, 

I´m trying to port [this](https://github.com/suchmememanyskill/CYD-Klipper) project from ESP32-2432S028R to ESP32-8048S070 and I can not make it compile.  Compiler give me this issue: 

`c:/users/dronecz/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-8048S070C/src/main.cpp.o:C:\Users\dronecz\Documents\GitHub\CYD-Klipper\CYD-Klipper/.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/include/esp32_smartdisplay.h:37: multiple definition of "touch_calibration_data"; .pio/build/esp328048S070C/src/core/screen_driver.cpp.o:C:\Users\dronecz\Documents\GitHub\CYD-Klipper\CYD-Klipper/.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/include/esp32_smartdisplay.h:37: first defined here
c:/users/dronecz/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-8048S070C/src/ui/ip_setup.cpp.o:C:\Users\dronecz\Documents\GitHub\CYD-Klipper\CYD-Klipper/.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/include/esp32_smartdisplay.h:37: multiple definition of "touch_calibration_data"; .pio/build/esp32-8048S070C/src/core/screen_driver.cpp.o:C:\Users\dronecz\Documents\GitHub\CYD-Klipper\CYD-Klipper/.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/include/esp32_smartdisplay.h:37: first defined here
c:/users/dronecz/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-8048S070C/src/ui/main_ui.cpp.o:C:\Users\dronecz\Documents\GitHub\CYD-Klipper\CYD-Klipper/.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/include/esp32_smartdisplay.h:37: multiple definition of "touch_calibration_data"; .pio/build/esp32-8048S070C/src/core/screen_driver.cpp.o:C:\Users\dronecz\Documents\GitHub\CYD-Klipper\CYD-Klipper/.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/include/esp32_smartdisplay.h:37: first defined here
c:/users/dronecz/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32-8048S070C/src/ui/panels/settings_panel.cpp.o:C:\Users\dronecz\Documents\GitHub\CYD-Klipper\CYD-Klipper/.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/include/esp32_smartdisplay.h:37: multiple definition of "touch_calibration_data"; .pio/build/esp32-8048S070C/src/core/screen_driver.cpp.o:C:\Users\dronecz\Documents\GitHub\CYD-Klipper\CYD-Klipper/.pio/libdeps/esp32-8048S070C/esp32_smartdisplay/include/esp32_smartdisplay.h:37: first defined here`

If I remove `"-DBOARD_HAS_TOUCH"` from JSON file, I can compile code and upload it to the display. Code does not work properly yet, but I can at least see something on the screen (if I comment out some lines in setup and loop of the program). 

[Here](https://github.com/dronecz/CYD-Klipper) is my fork with changes for the bigger display. 

Many thanks for any help and thank you for your awesome library!

rzeldent, 2024-01-21:
The smartdisplay.h file can only be included once; but is included in multiple projects:
- screen_driver.cpp
- settings_panel.cpp

Can you try to put extern in front of the definition in the smartdisplay.h? touch_calibration_data_t touch_calibration_data;

dronecz, 2024-01-22:
Thank you very much!

I added an extern to the esp32_smartdisplay.h and now I can compile it with touch. 
[image]

closed 2024-02-02 by rzeldent