rzeldent/esp32-smartdisplay #98: 8048S043C too slow and touch is not working

sstojos, 2024-01-21:
Hi, 

I like where this project is going. I managed to convert my very complex lvgl project very quickly and get compiled and running on 8048S043C board.

Unfortunately I found that config for board 8048S043C is too slow and CPU utilization is too high with this lib. 

I hope that it is only config that is not properly optimized and that it will be fixed at some point so that I can start using it at some point.

- maybe lvgl buffer config is too small - my buffer is 1/4 of total number of pixels - I don't know what is this lib default buffer size
- maybe it is because I am using a different driver that is faster. I am using moononournation/GFX Library for Arduino@1.3.7 - I don't know what  this library uses.

Also touch is not working. I did not have time to investigate why. I can see this library uses the correct chip GT911 for this board.

I use SPIFFS and this lib board config changed the definition from default to minimum. I had to uncomment this to get my already formatted SPIFFS mountable that uses default config.

I have copied bellow my code for this board. Maybe this can help to clarify if I can achieve the same speed with this lib.

Cheers.

my platformio.ini :

```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
board_upload.flash_size = 16MB
board_build.partitions = default_16MB.csv
build_flags = 
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
board_build.arduino.memory_type = qio_opi
lib_deps = 
	tamctec/TAMC_GT911@^1.0.2
	lvgl/lvgl@^8.3.9
	moononournation/GFX Library for Arduino@1.3.7
```

Here is how I initialize driver (GFX library).

```
/*******************************************************************************
 * Start of Arduino_GFX setting
 ******************************************************************************/
#define TFT_BL 2
#define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

// create a display bus object
Arduino_ESP32RGBPanel* bus = new Arduino_ESP32RGBPanel(
    40, 41, 39, 42,     /* DIS_DE, DIS_VSYNC, DIS_HSYNC, DIS_PCLK, */
    45, 48, 47, 21, 14, /* DIS_R_BUS, */
    5, 6, 7, 15, 16, 4, /* DIS_G_BUS, */
    8, 3, 46, 9, 1,     /* DIS_B_BUS, */
    0, 8, 4, 8,        /* DIS_HS_POL, DIS_HS_FP, DIS_HS_PW, DIS_HS_BP, */
    0, 8, 4, 8,        /* DIS_VS_POL, DIS_VS_FP, DIS_VS_PW, DIS_VS_BP, */
    1, 14000000 );        /* DIS_PC_A_N, DIS_SPEED ); */

// create a display driver object
Arduino_RGB_Display* gfx = new Arduino_RGB_Display( 800, 480, bus );

#endif /* !defined(DISPLAY_DEV_KIT) */
/*******************************************************************************
 * End of Arduino_GFX setting
 ******************************************************************************/
```

Here is how display flushing and touch read are implemented:

```
/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint32_t w = (area->x2 - area->x1 + 1);
  uint32_t h = (area->y2 - area->y1 + 1);

#if (LV_COLOR_16_SWAP != 0)
  gfx->draw16bitBeRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#else
  gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#endif

  lv_disp_flush_ready(disp);
}

void my_touchpad_read(lv_indev_drv_t *indev_driver, lv_indev_data_t *data)
{
  if (touch_has_signal())
  {
    if (touch_touched())
    {
      data->state = LV_INDEV_STATE_PR;

      /*Set the coordinates*/
      data->point.x = touch_last_x;
      data->point.y = touch_last_y;
    }
    else if (touch_released())
    {
      data->state = LV_INDEV_STATE_REL;
    }
  }
  else
  {
    data->state = LV_INDEV_STATE_REL;
  }
}
```

And finally here is how I initialize  lvgl :

```
  gfx->begin();
  pinMode(TFT_BL, OUTPUT);
  digitalWrite(TFT_BL, HIGH);

  lv_init();
  delay(10);
  touch_init();
  screenWidth = gfx->width();
  screenHeight = gfx->height();
   
  disp_draw_buf = (lv_color_t *)heap_caps_malloc(sizeof(lv_color_t) * screenWidth * screenHeight/4 , MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);

  if (!disp_draw_buf) {

    Log.error(F("LVGL disp_draw_buf allocate failed!"CR));

  } else {

    lv_disp_draw_buf_init(&draw_buf, disp_draw_buf, NULL, screenWidth * screenHeight/4);
    //    lv_disp_draw_buf_init( &draw_buf, buf, NULL, 800 * 480 / 4 );
    /* Initialize the display */
    lv_disp_drv_init(&disp_drv);
    /* Change the following line to your display resolution */
    disp_drv.hor_res = screenWidth;
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    disp =  lv_disp_drv_register(&disp_drv);

    /* Initialize the (dummy) input device driver */
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register(&indev_drv);

    Log.info(F("LVGL Display Setup done"CR));

    Log.trace(F("Initializing Screen"CR));

  }
```

rzeldent, 2024-01-21:
Hi

The buffersize is for now in the file smartdisplay.h, you can change this.

// LVGL lines buffered
#define LVGL_PIXEL_BUFFER_LINES 16

This is a project still in the startup phase and currently I'm the only one working on it. So suggestions and bugs are welcome.
There is still a lot to parametrize, like to Pickel clock (PCLK) (that's quite conservative) buffers etc...
Other libraries life GFX are marvels and kudo's for the people that keep them up to date and add new displays. However, It is full of trickery and optimizations that are difficult to understand.
I try to make a community to have these displays supported with as little effort as possible while maintaining a low memory footprint and standard drivers. This turns out to be difficult as there are multiple versions of the same boards out there.

You can take a look at the https://github.com/rzeldent/platformio-espressif32-sunton/tree/develop repository to see what needs to be configured. Maybe you can tweak your display....

rzeldent, 2024-01-21:
Hi Stojos,

Speed is indeed a bit slow on the 8048S043C. Increasing the pixel buffer does not help much. Increasing the PCLK creates glitching... 
And indeed GT911 does not function. Schematics are hard to find but think the I2C values are correct. Added to the todo list...

rzeldent, 2024-02-07:
Has been an improvement on the speed; quite an improvement using SPI ram. Take a look if this at least removes the speed impediment.  Is currently on the [develop](https://github.com/rzeldent/esp32-smartdisplay/tree/develop) branch

rzeldent, 2024-02-07:
The values for the I2C GT911 are correct. I looked at my board again and noticed it was a resistive version. So the touch test I did before was logical it did not work.

Compiling the -R variant worked immediately. Could you try to compile that one? Can you make a picture of the back of the board to check the touch version?

rzeldent, 2024-02-18:
Speed issue should be solved. SPI ram is used for S3 boards.

closed 2024-02-18 by rzeldent