rzeldent/esp32-smartdisplay #111: No display on ESP32-2432S032

sonictruth, 2024-01-25:
Hi,
I'm  getting this errors and also the screen is not updating.
```
E (166) spi_master: check_trans_valid(689): txdata transfer > host maximum
E (166) lcd_panel.io.spi: panel_io_spi_tx_color(380): spi transmit (queue) color failed
```

What could be the issue ? 
Thanks!

sonictruth, 2024-01-26:
The fix was to set:
```
    const spi_bus_config_t spi_bus_config = {
        .max_transfer_sz = LCD_WIDTH * LCD_HEIGHT,
```
in .pio/libdeps/esp32-2432S032C/esp32_smartdisplay/src/lvgl_st7789.c

rzeldent, 2024-01-26:
Retested (without adding a max_transfer_sz) and works.
Strange, the max_transfer_sz  is never defined in SPI. Are using a large pixel buffer?

rzeldent, 2024-01-26:
Anyone else with similar problems?

balintbuchert, 2024-01-28:
> Anyone else with similar problems?

Yes I run in to the same issue with a esp32-3248S035C board.

ab-413, 2024-02-15:
> Anyone else with similar problems?

Yes , same issue with a esp32-2432S028R board.

Following @sonictruth comment I did the same modification in:
.pio\libdeps\esp32-2432S028R\esp32_smartdisplay\src\lvgl_ili9341_spi.c

rzeldent, 2024-02-15:
So, setting the .max_transfer_sz helped?

ab-413, 2024-02-16:
> So, setting the .max_transfer_sz helped?

Yes, display works fine

closed 2024-04-16 by rzeldent

silvelo, 2025-05-28:
Same error on esp32-2432S028Rv3 and change esp32-2432S028Rv3.json:

"'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT/4)'",
to 
"'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT/100)'",

And now works well