rzeldent/esp32-smartdisplay #121: Demo will not start in ESP32-3248S035

johncharris, 2024-02-05:
When trying to start this device I see a screen with static, then the display just goes white and restarts. The Serial Monitor shows the crash below. I tried the regular and develop branches of esp32-smartdisplay library.

I was able to load this on an ESP32-2432S028R without issue.

Thank you for your help.

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    21][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   294][I][main.cpp:26] setup(): CPU: ESP32-D0WD-V3 rev3, CPU Freq: 240 Mhz, 2 core(s)
[   295][I][main.cpp:27] setup(): Free heap: 287616 bytes
[   296][I][main.cpp:28] setup(): Free PSRAM: 0 bytes
[   301][I][main.cpp:29] setup(): SDK version: v4.4.6-dirty
[   306][D][esp32_smartdisplay.c:209] smartdisplay_init(): smartdisplay_init
[   315][D][lvgl_st7796_spi.c:33] lvgl_lcd_init(): lvgl_lcd_init
[   546][D][esp32_smartdisplay.c:109] smartdisplay_lcd_set_backlight(): smartdisplay_lcd_set_backlight. duty:0
[   546][D][lvgl_gt911_i2c.c:167] lvgl_touch_init(): lvgl_touch_init
[   551][D][lvgl_gt911_i2c.c:179] lvgl_touch_init(): i2c_param_config. host: 0
[   558][D][lvgl_gt911_i2c.c:181] lvgl_touch_init(): i2c_driver_install host: 0
[   565][D][lvgl_gt911_i2c.c:197] lvgl_touch_init(): esp_lcd_new_panel_io_i2c. host: 0
E (554) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400891e8
file: ".pio/libdeps/esp32-3248S035C/esp32_smartdisplay/src/lvgl_gt911_i2c.c" line 200
func: lvgl_touch_init
expression: esp_lcd_panel_io_rx_param(io_handle, 0x8140, &gt_info, sizeof(struct GTInfo))

abort() was called at PC 0x400891eb on core 1

Backtrace: 0x40083ee5:0x3ffb2090 0x400891f5:0x3ffb20b0 0x4008ec0d:0x3ffb20d0 0x400891eb:0x3ffb2150 0x400f80db:0x3ffb2170 0x400f7c90:0x3ffb2230 0x400d2392:0x3ffb2250 0x4011e972:0x3ffb2290

  #0  0x40083ee5:0x3ffb2090 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x400891f5:0x3ffb20b0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x4008ec0d:0x3ffb20d0 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x400891eb:0x3ffb2150 in _esp_error_check_failed at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_err.c:43
  #4  0x400f80db:0x3ffb2170 in lvgl_touch_init at .pio/libdeps/esp32-3248S035C/esp32_smartdisplay/src/lvgl_gt911_i2c.c:180 (discriminator 1)
  #5  0x400f7c90:0x3ffb2230 in smartdisplay_init at .pio/libdeps/esp32-3248S035C/esp32_smartdisplay/src/esp32_smartdisplay.c:268
  #6  0x400d2392:0x3ffb2250 in setup() at src/main.cpp:45
  #7  0x4011e972:0x3ffb2290 in loopTask(void*) at /home/john/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: c480e07960caf7a1
```

johncharris, 2024-02-07:
Looks like it was an issue with the touch panel not being properly attached. After fiddling around with the FFC for the touch pad, both my 3.5" displays are working.

closed 2024-02-07 by johncharris

rzeldent, 2024-02-07:
Glad you found it!
BTW This is not a crash but an assertion that something is not right!