rzeldent/esp32-smartdisplay #122: configurable display draw buffer support

PingguSoft, 2024-02-06:
display update is so slow for boards with big resolution.
it is caused by LVGL_PIXEL_BUFFER_LINES is limited to 16.
I think it is better to give options such as SPIRAM and lines for drawBuffer.

With esp32-4848S040CIY1 board, below changes show fast display update.

```
void smartdisplay_init()
{
<snip>
//   uint drawBufferPixels = LCD_WIDTH * LVGL_PIXEL_BUFFER_LINES;
//   void *drawBuffer = heap_caps_malloc(sizeof(lv_color16_t) * drawBufferPixels, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);

  uint drawBufferPixels = LCD_WIDTH * LCD_HEIGHT;
  void *drawBuffer = heap_caps_malloc(sizeof(lv_color16_t) * drawBufferPixels, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
```

thanks.

rzeldent, 2024-02-07:
Excellent suggestion. Focus was on stability and get the displays working but this indeed should make it much more usable!

rzeldent, 2024-02-07:
Is now in [develop](https://github.com/rzeldent/esp32-smartdisplay/tree/develop)

Also did this for other boards, let's hope it works as well as the esp32-8048S070 (that was really slow)
Thanks again,

closed 2024-02-07 by rzeldent