rzeldent/esp32-smartdisplay #129: ESP32-2424S012N/C blank display

TheCooolNerd, 2024-02-11:
Hi, 

I have the 1.2 inch round display. 
The code uploads fine with no errors. I can see the backlight light up but nothing on the screen. 
The serial monitor just shows this :

`14:49:40.777 -> ESP-ROM:esp32c3-api1-20210207
14:49:40.777 -> Build:Feb  7 2021
14:49:40.777 -> rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
14:49:40.777 -> Saved PC:0x4206bcca
14:49:40.777 -> SPIWP:0xee
14:49:40.777 -> mode:DIO, clock div:1
14:49:40.777 -> load:0x3fcd5810,len:0x438
14:49:40.777 -> load:0x403cc710,len:0x90c
14:49:40.777 -> load:0x403ce710,len:0x25f4
14:49:40.777 -> entry 0x403cc710`

The code is running as i can see the board connect to the wifi network in the listed devices. 

Dont know how to begin trouble shooting.

rzeldent, 2024-02-11:
Hi Coool nerd,

What version are you using?

TheCooolNerd, 2024-02-11:
Library : Latest
Demo : latest
All downloaded within the span of last week.

rzeldent, 2024-02-11:
There has been a release yesterday so doubt you have the latest version if you downloaded this last week.

TheCooolNerd, 2024-02-11:
How can i check/update the library/demo version?

rzeldent, 2024-02-11:
guess the simplest is to clone it in a new direcory

TheCooolNerd, 2024-02-11:
cloned the demo to a new directory with the submodules
selected my board in the platformio.ini by removing the # 
uploaded the project.
same issue. backlight turns on, black display. 
same output on the serial monitor

rzeldent, 2024-02-11:
Retested and confirmed. For now use an older version. Wil take a look at it. You can also compare it to previous versions.

TheCooolNerd, 2024-02-11:
Thank you. will compare and see if any changes make sense to me. 
for now the only thing that pops is that even though the flash mode is qio, the serial output says DIO. may be irrelevant as well.

rzeldent, 2024-02-11:
Found typo introduced during refactoring. You can switch to the develop branch or apply https://github.com/rzeldent/esp32-smartdisplay/commit/bb656fd383c51bde94e531549db0f0f0b2646bc4

NB boards are also updated to have logging output at boot

Enjoy!

[image]

closed 2024-02-11 by rzeldent

rzeldent, 2024-02-11:
Thanks for reporting the issue!