rzeldent/esp32-smartdisplay #130: 8048S070C Backlight does not turn on

LunarFlames, 2024-02-11:
Version: esp32-smartdisplay 2.0.5
Board: 8048S070C v1.3
Board Definition: esp32-8048S070C.json

The backlight does not turn on for this board when I try to compile [esp32-smartdisplay-demo](https://github.com/rzeldent/esp32-smartdisplay-demo). If I use a flashlight, I can see the demo is running as expected.

I have to go to **esp32_smartdisplay.h** and modify the following start at line 8. I pulled these numbers from the 7.0inch_ESP32-8048S070 demo package. I'm not sure if they're a correct values because the demo package also had the red and blue GPIO pins swapped. PWM_BITS_BCKL was already set to 8.
```
#define PWM_CHANNEL_BCKL 0
#define PWM_FREQ_BCKL 300
#define PWM_BITS_BCKL 8
```

When I compile it again, the backlight turns on and I can see the demo. 

I have not tried to change the brightness once it's initialized.

I'm not sure if this is supposed to be set somewhere else. I have checked the board definition but I don't immediately see a flag that would set this option.

GilchristT, 2024-02-12:
I don't have the same board to test but I ran into a similar issue with an esp32-4848S040

Looking at the boot logs from the serial output I noticed 

```
[   411][D][esp32_smartdisplay.c:109] smartdisplay_lcd_set_backlight(): smartdisplay_lcd_set_backlight. duty:0.500000
```

I found setting the backlight to 1 after ```smartdisplay_init()```  did the trick for me.

```cpp
void setup()
{
  smartdisplay_init();
  smartdisplay_lcd_set_backlight(1);

  auto disp = lv_disp_get_default();
  ...
```

Hope this helps.

p.s. Thanks to @rzeldent for the great work on this library.

rzeldent, 2024-02-12:
Maybe this is related? https://github.com/rzeldent/esp32-smartdisplay-demo/issues/21

rzeldent, 2024-02-18:
The backlight is controlled using a the GPIO_BCKL (develop) or BCKL (main) pin.
This pin is attached to the PWM channel so it can be modulated. It might be that the PWM channel is not correct or attached or used by another peripheral.
The easiest way is indeed to just use a digitalwrite(GPIO_BCKL, true) so it will be at 100% brightness...

Would however be interesting to know why this does not work on some boards. I tested mine and cannot reproduce this. Some hints are:
- The PWM routines are changed with the new Espressif release
- recently frequency has been increased because before it was audible on some boards
- changes in definitions?

Hope this goves some insights in troubleshooting. Please let me know if somethings pops up or is not correct!

werde45, 2024-03-15:
It seems the display has other backlight pins. I bought some of them last year 2023 and now some in february 2024. If it try to set backlight it turns to black.

rzeldent, 2024-03-20:
Hi Werde,

That's possible, they do seem to change a bit the schematics. Some time ago I supported 2 versions of the esp32-8048S070 but one was dropped. Take a look in the Sunton defines to find the pins!
If it works for you I can add (again) a new definition...