rzeldent/esp32-smartdisplay #132: ESP32-2432S022C - Touch initialization problems

Nostromo0, 2024-02-14:
Hello there,
I'm having an issue with running the demo project with a ESP32-2432S022C board:

```
E (463) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (469) CST816S: read_id(176): I2C read failed
E (474) CST816S: esp_lcd_touch_new_i2c_cst816s(86): Read version failed
E (480) CST816S: Initialization failed!
```
This happens on `lvgl_touch_init`

Any idea what might be the problem?
Thanks.

Nostromo0, 2024-02-14:
I also tried to use the board's factory sample project using the Arduino IDE but it gave me a I2C timeout error

rzeldent, 2024-02-14:
Do you have a photo of the board? To what controller is present (or none)...

Nostromo0, 2024-02-15:
I'll send you pictures as soon as i can.
Right now i can only give you the link i bought the board from:
https://aliexpress.com/item/1005006102451859.html

Nostromo0, 2024-02-15:
Here they are:
[image]

[image]

rzeldent, 2024-02-15:
Thanks!

IO think the U6: 
[image]

Is an xpt2046 so resistive...

rzeldent, 2024-02-22:
Assume this was the issue?

Nostromo0, 2024-02-26:
Hello there, it appears that i have been using the wrong model the whole time when compiling the project.
I'm sorry if i may have caused you to waste any time. I'm also gonna go ahead and close the issue.
Thanks for developing this amazing library!

closed 2024-02-26 by Nostromo0