rzeldent/esp32-smartdisplay #134: ESP32-2432S028v2 not working anymore

JoDaNl, 2024-02-23:
Hi,

In issue https://github.com/rzeldent/esp32-smartdisplay/issues/59 I reported that ESP32-2432S028 was not working i.e., only showed a fully white screen. This was due to adiffrent initialisation & solved by providing the v2 board version with a different vendor init sequence.

I now moved my project to the latest state of this (great) library and I get a white screen again.
So I did some debugging and found out that the vendor sprcific init cmd's (which are in the board-file) are not loaded correctly anymore.

I looked into the 'esp_lcd_ili9341.c' file and added some debug print statements:

`    // Create ili9341 panel handle
    const esp_lcd_panel_dev_config_t panel_dev_config = {
        .reset_gpio_num = ILI9341_DEV_CONFIG_RESET_GPIO_NUM,
        .color_space = ILI9341_DEV_CONFIG_COLOR_SPACE,
        .bits_per_pixel = ILI9341_DEV_CONFIG_BITS_PER_PIXEL,
        .flags = {
            .reset_active_high = ILI9341_DEV_CONFIG_FLAGS_RESET_ACTIVE_HIGH},
        .vendor_config = ILI9341_DEV_CONFIG_VENDOR_CONFIG};
    if (panel_dev_config.vendor_config)
    {
        log_d("Initialization with vendor config");

        ili9341_vendor_config_t *c1, *c2, *c3;

        c1 = (ili9341_vendor_config_t *)panel_dev_config.vendor_config;
        printf("DBG1: init_cmds_size=%d\n",c1->init_cmds_size);
        c2 = (ili9341_vendor_config_t *)ILI9341_DEV_CONFIG_VENDOR_CONFIG;

        printf("DBG2: init_cmds_size=%d\n",c2->init_cmds_size);
        c3 = (ili9341_vendor_config_t[]){{.init_cmds=(ili9341_lcd_init_cmd_t[]){{.cmd=0xCF,.data=(uint8_t[]){0x00,0xC1,0x30},.data_bytes=3},{.cmd=0xED,.data=(uint8_t[]){0x64,0x03,0x12,0x81},.data_bytes=4},{.cmd=0xE8,.data=(uint8_t[]){0x85,0x00,0x78},.data_bytes=3},{.cmd=0xCB,.data=(uint8_t[]){0x39,0x2C,0x00,0x34,0x02},.data_bytes=5},{.cmd=0xF7,.data=(uint8_t[]){0x20},.data_bytes=1},{.cmd=0xEA,.data=(uint8_t[]){0x00,0x00},.data_bytes=2},{.cmd=0xC0,.data=(uint8_t[]){0x10},.data_bytes=1},{.cmd=0xC1,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0xC5,.data=(uint8_t[]){0x30,0x30},.data_bytes=2,},{.cmd=0xC7,.data=(uint8_t[]){0xB7},.data_bytes=1},{.cmd=0x3A,.data=(uint8_t[]){0x55},.data_bytes=1},{.cmd=0x36,.data=(uint8_t[]){0x08},.data_bytes=1},{.cmd=0xB1,.data=(uint8_t[]){0x00,0x1A},.data_bytes=2},{.cmd=0xB6,.data=(uint8_t[]){0x08,0x82,0x27},.data_bytes=3},{.cmd=0xF2,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0x26,.data=(uint8_t[]){0x01},.data_bytes=1},{.cmd=0xE0,.data=(uint8_t[]){0x0F,0x2A,0x28,0x08,0x0E,0x08,0x54,0xA9,0x43,0x0A,0x0F,0x00,0x00,0x00,0x00},.data_bytes=15},{.cmd=0xE1,.data=(uint8_t[]){0x00,0x15,0x17,0x07,0x11,0x06,0x2B,0x56,0x3C,0x05,0x10,0x0F,0x3F,0x3F,0x0F},.data_bytes=15},{.cmd=0x2B,.data=(uint8_t[]){0x00,0x00,0x01,0x3F},.data_bytes=4},{.cmd=0x2A,.data=(uint8_t[]){0x00,0x00,0x00,0xEF},.data_bytes=4},{.cmd=0x21},{.cmd=0x11,.delay_ms=120},{.cmd=0x29,.delay_ms=1}},.init_cmds_size=23}};
        printf("DBG2: init_cmds_size=%d\n",c3->init_cmds_size);

    }
`

So just wanted to check the nr of cmd provided.
I get the followin output:

`[   243][D][esp32_smartdisplay.c:209] smartdisplay_init(): smartdisplay_init
[   246][D][lvgl_ili9341_spi.c:33] lvgl_lcd_init(): lvgl_lcd_init
[   250][D][lvgl_ili9341_spi.c:79] lvgl_lcd_init(): Initialization with vendor config
DBG1: init_cmds_size=13113
DBG2: init_cmds_size=13113
DBG2: init_cmds_size=23`

Not sure what's causing this...but the define ILI9341_DEV_CONFIG_VENDOR_CONFIG is somehow not landing correctling in the panel_dev_config  struct.

JoDaNl, 2024-02-28:
Fixed this by removing escapes in the boards\esp32-2432S028Rv2.json file.
Changed the ILI9341_DEV_CONFIG_VENDOR_CONFIG line accoring to below:

`"'-D ILI9341_DEV_CONFIG_VENDOR_CONFIG=(ili9341_vendor_config_t[]){{.init_cmds=(ili9341_lcd_init_cmd_t[]){{.cmd=0xCF,.data=(uint8_t[]){0x00,0xC1,0x30},.data_bytes=3},{.cmd=0xED,.data=(uint8_t[]){0x64,0x03,0x12,0x81},.data_bytes=4},{.cmd=0xE8,.data=(uint8_t[]){0x85,0x00,0x78},.data_bytes=3},{.cmd=0xCB,.data=(uint8_t[]){0x39,0x2C,0x00,0x34,0x02},.data_bytes=5},{.cmd=0xF7,.data=(uint8_t[]){0x20},.data_bytes=1},{.cmd=0xEA,.data=(uint8_t[]){0x00,0x00},.data_bytes=2},{.cmd=0xC0,.data=(uint8_t[]){0x10},.data_bytes=1},{.cmd=0xC1,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0xC5,.data=(uint8_t[]){0x30,0x30},.data_bytes=2,},{.cmd=0xC7,.data=(uint8_t[]){0xB7},.data_bytes=1},{.cmd=0x3A,.data=(uint8_t[]){0x55},.data_bytes=1},{.cmd=0x36,.data=(uint8_t[]){0x08},.data_bytes=1},{.cmd=0xB1,.data=(uint8_t[]){0x00,0x1A},.data_bytes=2},{.cmd=0xB6,.data=(uint8_t[]){0x08,0x82,0x27},.data_bytes=3},{.cmd=0xF2,.data=(uint8_t[]){0x00},.data_bytes=1},{.cmd=0x26,.data=(uint8_t[]){0x01},.data_bytes=1},{.cmd=0xE0,.data=(uint8_t[]){0x0F,0x2A,0x28,0x08,0x0E,0x08,0x54,0xA9,0x43,0x0A,0x0F,0x00,0x00,0x00,0x00},.data_bytes=15},{.cmd=0xE1,.data=(uint8_t[]){0x00,0x15,0x17,0x07,0x11,0x06,0x2B,0x56,0x3C,0x05,0x10,0x0F,0x3F,0x3F,0x0F},.data_bytes=15},{.cmd=0x2B,.data=(uint8_t[]){0x00,0x00,0x01,0x3F},.data_bytes=4},{.cmd=0x2A,.data=(uint8_t[]){0x00,0x00,0x00,0xEF},.data_bytes=4},{.cmd=0x21},{.cmd=0x11,.delay_ms=120},{.cmd=0x29,.delay_ms=1}},.init_cmds_size=23}}'",`

Beleive (have not proven this) that the initialization code-sequence was interpreted as a string...which was assigned to a void *...which the compiler accepted.

rzeldent, 2024-02-28:
That looks like a good find!
Will update this!

rzeldent, 2024-02-28:
Updated in main and develop

closed 2024-03-20 by rzeldent