rzeldent/esp32-smartdisplay #135: Display rotates but not the touch sensor

AllanOricil, 2024-02-23:
## Expected behaviour
Screen's 1 button when moves to the right screen. While screen's 2 button moves to the left screen.

[image]

[image]

[image]

[image]

## Actual behaviour

The screen is rotated, but to touch the button I have to think that it has been "drawn" when the screen isn't rotate. What am I doing wrong? 

## Additional Information

`platform.io`
````ini
[env:esp32-2432S028Rv3]
platform = espressif32
board = esp32-2432S028Rv3
framework = arduino
lib_deps = 
    https://github.com/rzeldent/esp32-smartdisplay.git
build_flags =
    -Ofast
    -Wall
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE
    # LVGL settings. Point to your lv_conf.h file
    -D LV_CONF_PATH="${PROJECT_DIR}/lv_conf.h"
````

`main.cpp`

````c
#include <Arduino.h>
#include <esp32_smartdisplay.h>
#include <ui/ui.h>

void setup() {
  smartdisplay_init();

  auto disp = lv_disp_get_default();
  lv_disp_set_rotation(disp, LV_DISP_ROT_90);
  ui_init();
}

void loop() {
  lv_timer_handler();
}

````

rzeldent, 2024-02-23:
Hi Allan,

Need to investigate this further. Looks like the functions 
```
esp_lcd_touch_set_swap_xy
esp_lcd_touch_set_mirror_x
esp_lcd_touch_set_mirror_y
```
 are not implemented in the drivers....

AllanOricil, 2024-02-23:
@rzeldent thanks for the quick reply. I'm a noob in c and embedded stuff so I don't really know how to fix. If you can't the issue let me know what other information you need. I can make a video for you.

rzeldent, 2024-02-23:
Hi Allan,

No thanks but should be able to reproduce it myself so know when it is fixed...

xperiments, 2024-02-24:
Same here, for a temp "dev" while fixed... I changed the       
"'-D TOUCH_SWAP_XY=true'"
while using
lv_disp_set_rotation(disp, LV_DISP_ROT_90);

rzeldent, 2024-02-25:
Yes, looking into this. Looks like the espressif drivers do not take this into account.
Working on it...

rzeldent, 2024-02-25:
Hi Allan/xperiments,

A quick update...
I made a new xpt2046 driver. This is available in the develop branch but think it is ready to be tested.

AllanOricil, 2024-02-25:
> his is available in the develop branch but think it is ready to be tested.

@rzeldent could you guide me to what I have to do? Do i need to download a new board and add to platform.io? do I need to then change board and [env:...] in platformio.ini? What else?

thanks

rzeldent, 2024-02-25:
just type:

```
git checkout develop
```

This should change the branch to develop.

AllanOricil, 2024-02-25:
Like this?  It seems to have installed 5b0bbe7
[image]

rzeldent, 2024-02-25:
Yes!

The new xpt2046 library is loaded (see before last line). Remember; might work but still working on it...

AllanOricil, 2024-02-25:
@rzeldent it is printing lot of things to the serial haha

[image]

but it seemed to have installed without a problem

rzeldent, 2024-02-25:
Yes, welcome to the world of debugging!
This can be disabled by changing the  debug level to

-DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN

xperiments, 2024-02-25:
Not working here with the develop branch... when rotating to 90

THX!!!

rzeldent, 2024-02-25:
Thank you for your support...

AllanOricil, 2024-02-26:
@xperiments @rzeldent 

for now I think I won't need buttons for this pet project: https://github.com/AllanOricil/esp32-mfa-totp-generator
But in the future, when I have too many registered MFA secrets that can't fit on a single screen, I will need to add some buttons in the screen, while still having its UI rotated 90 degrees.

AllanOricil, 2024-02-26:
@rzeldent do you have an example where I can create components and add them to the screen dynamically in a "flex layout". I could not find a way to do it on that app called `Squareline_Studio`

AllanOricil, 2024-03-03:
@rzeldent I reached a point in my personal project which requires toutch to work properly when rotating the device. At the moment, because of this bug, in order to see the TOTP codes that are outside of the viewport I need to scroll horizontally if I want to move up or down haha

https://github.com/AllanOricil/esp32-mfa-totp-generator

rzeldent, 2024-03-03:
HI Allan,

I made some adjustments to the devlop branch: https://github.com/rzeldent/esp32-smartdisplay/commit/38956edb0aac544fb725ded3eee3196fbad8223a

You could test that (haven't got the time)...

AllanOricil, 2024-03-03:
@rzeldent I need your help with other subject. I hope you can help me. How can I flash this model of esp32? I want to store a key in it. I also want to enable encryption so that it's flashed mem is encrypted and protected.

rzeldent, 2024-03-03:
Hi Allan,

Securing is tricky but can be done using the Espressif IDF, theire native development environment.
You can find an explanation here: https://www.motius.com/post/how-to-build-a-secure-iot-prototype-with-arduino-and-esp32

Optionally, just store the encrypted key in the code and use some runtime decryption to get the key. This makes it only more difficult but for practical security should be enough?!

tadelv, 2024-03-03:
hello all, just chiming in - I have a 2432S028Rv3 and using a `DISP_ROT_270`. I was able to get it to work (on `develop` branch for both boards and the library) with the following mods to the board `.json` file:
```diff
diff --git a/esp32-2432S028Rv3.json b/esp32-2432S028Rv3.json
--- a/esp32-2432S028Rv3.json
+++ b/esp32-2432S028Rv3.json
@@ -72,6 +72,6 @@
       "'-D XPT2046_TOUCH_CONFIG_LEVELS_RESET=0'",
       "'-D XPT2046_TOUCH_CONFIG_LEVELS_INTERRUPT=0'",
-      "'-D TOUCH_SWAP_XY=false'",
-      "'-D TOUCH_SWAP_X=true'",
+      "'-D TOUCH_SWAP_XY=true'",
+      "'-D TOUCH_SWAP_X=false'",
       "'-D TOUCH_SWAP_Y=false'",
       "'-D BOARD_HAS_TF'",

```

Hope this is relevant 😅

AllanOricil, 2024-03-03:
@rzeldent I can't build with `develop`

I did a `Full Clean` then `Build` and got this error

[image]

AllanOricil, 2024-03-03:
@rzeldent `38956edb0aac544fb725ded3eee3196fbad8223a` WORKS!

AllanOricil, 2024-03-03:
BEHOLD!

https://github.com/AllanOricil/esp32-mfa-totp-generator/tree/v0.2.2

the-real-Isual, 2024-03-07:
I have a similar issue. I used the bare minimum sketch (https://github.com/rzeldent/esp32-smartdisplay-demo) and a 2432S028  V3, and the commit number mentioned by Allan, but the touch input is still unchanged at 90°.

AllanOricil, 2024-03-08:
> I have a similar issue. I used the bare minimum sketch (https://github.com/rzeldent/esp32-smartdisplay-demo) and a 2432S028  V3, and the commit number mentioned by Allan, but the touch input is still unchanged at 90°. 

Make sure you also rotated the display using smartdisplay, and also don't forget to do a Full Clean in platformio after changing the dependency hash.

AllanOricil, 2024-03-11:
@rzeldent I noticed that just by switching from v3 to v2, I get a white screen. I switched to v2, did a full clean, then build and upload and monitor. When the UI loads, a white screen appears. Then I switched back to v3, repeated the same steps, and the UI was fine.

rzeldent, 2024-03-31:
Has been resolved.

closed 2024-03-31 by rzeldent