rzeldent/esp32-smartdisplay #138: How to enable 3rd party libraries  ( QR code , Barcode ) , based on the example project ?

andersondeoliveiramachado, 2024-03-02:
I used the example project to run an interface generated with GUI-GUIDER ( https://www.nxp.com/design/design-center/software/development-software/gui-guider:GUI-GUIDER )

[image]
[image]
[image]

Example running but with the qrcode , barcode components disabled.

[image]

\include\lv_conf.h 
`
/*QR code library*/
#define LV_USE_QRCODE 1
`

main.cpp
`#include <Arduino.h>
#include <WiFi.h>

#include <esp32_smartdisplay.h>
#include <generated/gui_guider.h>

#define WIFI_SSID "NETWORK"
#define WIFI_PASSWORD "12345678"

lv_ui guider_ui;

void setup()
{
#ifdef ARDUINO_USB_CDC_ON_BOOT    
    delay(5000);
#endif    
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    log_i("Board: %s", BOARD_NAME);
    log_i("CPU: %s rev%d, CPU Freq: %d Mhz, %d core(s)", ESP.getChipModel(), ESP.getChipRevision(), getCpuFrequencyMhz(), ESP.getChipCores());
    log_i("Free heap: %d bytes", ESP.getFreeHeap());
    log_i("Free PSRAM: %d bytes", ESP.getPsramSize());
    log_i("SDK version: %s", ESP.getSdkVersion());

    // Connect to WiFi
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    if (WiFi.waitForConnectResult() == WL_CONNECTED)
    {
    }

    smartdisplay_init();

    auto disp = lv_disp_get_default();
    // lv_disp_set_rotation(disp, LV_DISP_ROT_90);
    // lv_disp_set_rotation(disp, LV_DISP_ROT_180);
    // lv_disp_set_rotation(disp, LV_DISP_ROT_270);

    // call of the interface generated by the guide
    setup_ui(&guider_ui);

}

ulong next_millis;

void loop()
{
    auto const now = millis();
    if (now > next_millis)
    {
        next_millis = now + 500;
        char text_buffer[32];
        sprintf(text_buffer, "%lu", now);
    }
    lv_timer_handler();
}
`
The example simulates a price tag, and uses the 3rd party libraries ( QR code , Barcode )
https://docs.lvgl.io/master/libs/barcode.html
https://docs.lvgl.io/master/libs/qrcode.html
I was unable to enable the libraries.

[image]

rzeldent, 2024-03-02:
https://docs.lvgl.io/master/libs/qrcode.html

andersondeoliveiramachado, 2024-03-02:
I managed to solve it by manually placing the extra libs in the lvgl directory.

[image]

Is there any way to put it in plataromio.ini so that an extra lib is downloaded along with esp32-smartdisplay?

[image]

rzeldent, 2024-03-17:
Take a look at PR#30: https://github.com/rzeldent/esp32-smartdisplay-demo/pull/30/files

Just change define in the lv_conf.h file, include the headers and it works...

Updated the default demo with a QR code...

closed 2024-03-17 by rzeldent