rzeldent/esp32-smartdisplay #143: ESP32-2432S028 - Can't use SD card after initializing the UI

AllanOricil, 2024-03-12:
[image]

AllanOricil, 2024-03-12:
I further isolated the issue to `smartdisplay_init()`. After calling it, I can no longer use the sd card

[image]

rzeldent, 2024-03-12:
Hi Allan,

I think it is caused by the SPI bus using the same SPI bus as the SD card.

In the file esp32-2432S028R.json you can trt to change the SPI host to : "'-D ILI9341_SPI_HOST=SPI2_HOST'"

Choices are:

    SPI1_HOST=0,    ///< SPI1
    SPI2_HOST=1,    ///< SPI2
#if SOC_SPI_PERIPH_NUM > 2
    SPI3_HOST=2,    ///< SPI3

Or change the SPI HOST for the SD card....

AllanOricil, 2024-03-12:
> ILI9341_SPI_HOST

@rzeldent 
I'm using `esp32-2432S028Rv3.json` and there is no `-D ILI9341_SPI_HOST=SPI2_HOST` in extra_flags.

rzeldent, 2024-03-12:
Yes, then try the  "'-D ST7789_SPI_HOST=SPI2_HOST'" flag!

AllanOricil, 2024-03-12:
But it was already set with SPI2_HOST

rzeldent, 2024-03-12:
Yes, that is how it is now. But try to change this to SPI1_HOST or even SPI3_HOST....
The CS Card can also work in 2 modes, SPI or a 4 bits protocol. https://esp32.com/viewtopic.php?t=110

AllanOricil, 2024-03-12:
None worked.

I changed `-D ST7789_SPI_HOST=SPI2_HOST` to `-D ST7789_SPI_HOST=SPI3_HOST` at `~/.platformio/platforms/espressif32/boards`, full clean, build, upload and monitor and got the same result

I changed `-D ST7789_SPI_HOST=SPI2_HOST` to `-D ST7789_SPI_HOST=SPI1_HOST` at `~/.platformio/platforms/espressif32/boards`, then did a full clean, build, upload and monitor and got the same result

AllanOricil, 2024-03-12:
I also noticed that `SPI3_HOST` is already being used by the XPT2046 in all revisions of `esp32-2432S028R.json`

````json
"'-D XPT2046_SPI_HOST=SPI3_HOST'",
````

AllanOricil, 2024-03-12:
I just tried to free the SPI2_HOST and SPI3_HOST to let the sd card use them, and got the same result. Maybe I did it wrong?

````c
// all the following operations work
create_dir(SD, "/test3");
write_file(SD, "/test3/test.txt", "bla");
read_file(SD, "/test3/test.txt");

// SETUP UI
smartdisplay_init();
  
pinMode(TF_CS, OUTPUT);
pinMode(ST7789_SPI_CONFIG_CS_GPIO_NUM, OUTPUT);

digitalWrite(TF_CS, LOW); //enable SD card => option as it was already HIGH in the setup
digitalWrite(ST7789_SPI_CONFIG_CS_GPIO_NUM, HIGH); =>

// THIS FAILS even when disabled touch or lcd CS_GPIO pins
read_file(SD, "/test3/test.txt");

````

AllanOricil, 2024-03-12:
@rzeldent 
I changed both lcd and touch to use SPI2_HOST and then set the sd card to use SPI3_HOST and it worked. It was the only thing I could do to make it work. I also tried changign both to SPI3_HOST, and le tthe SD card using SPI2_HOST, but this setup did not work. However, I can no longer use the touch, and I need it. So, I think you have to create some sort of multiplexing to allow both touch and screen share the same SPI host. At least, that was what chat gpt told me. Is he right?

AllanOricil, 2024-03-12:
I have also tried this combination, and my code breaks at runtime. Apprentlly SPI1_HOST can only be used to flash.

````json
"'-D ST7789_SPI_HOST=SPI2_HOST'",
"'-D XPT2046_SPI_HOST=SPI1_HOST'",
"'-D TF_SPI_HOST=SPI3_HOST'",
````

AllanOricil, 2024-03-12:
I also tried this chatgpt suggestion. Put both touch and SD card on the same SPI3_HOST, then used the following code when trying to interact with the sd card. And again, it did not work.

````c
digitalWrite(XPT2046_SPI_CONFIG_CS_GPIO_NUM, HIGH); // Deselect touch sensor 
mySPI.beginTransaction(SPISettings(SPI_CLOCK_DIV2, MSBFIRST, SPI_MODE0));
digitalWrite(TF_CS, LOW); // Select SD card
read_file(SD, "/test3/test.txt");
// Further file operations
digitalWrite(TF_CS, HIGH); // Deselect SD card
mySPI.endTransaction();
digitalWrite(XPT2046_SPI_CONFIG_CS_GPIO_NUM, LOW);
````

AllanOricil, 2024-03-12:
@rzeldent 
Im blocked until this issue could be resolved. I tried everything I could have asked to chatgpt. I ran out of questions D:

AllanOricil, 2024-03-12:
According to chat gpt I have no option. This board was created without support for using SD card while touch and lcd are using the available spi buses. If you find a solution @rzeldent I would appreciate. I don't believe chat gpt is fully right.

AllanOricil, 2024-03-12:
Or could you make your lib handle both touch and screen on the same spi bus?

AllanOricil, 2024-03-12:
@rzeldent 
This video shows an implementation which uses a SPI Semaphore to control which peripherals can use the the spi bus. Maybe you can do the same for your lib, and expose this as a global spi semaphore which people can use to register their peripherals with ease. 

https://m.youtube.com/watch?v=8BU07bUQKvg

AllanOricil, 2024-03-12:
https://randomnerdtutorials.com/esp32-spi-communication-arduino/

AllanOricil, 2024-03-12:
This did not work as well. Both LCD and SD on the same SPI bus (SPI2_HOST). 

[image]

rzeldent, 2024-03-12:
Hi Allan,

The TFT_CS and ST7789_SPI_CONFIG_CS_GPIO_NUM should indeed do the multiplexing. CS stands for Chip Select.

Did you call the SD.begin(cspin) with the cspin specified? This should be handled automatically....

AllanOricil, 2024-03-12:
@rzeldent I did. Look at the last picture. 

Could you write a minimal example demonstrate that it endeed works? I could not make one.
Then you can share here with everyone.

AllanOricil, 2024-03-12:
Look, it does not work. This function called `add_new_secret` is called whenever a mqtt message is received. 

[image]

Obs: disregard the name of the function.
Obs: I'm just trying to read the file in that function because I did it without a problem in the setup, before calling  before calling `smartdisplay_init()`

rzeldent, 2024-03-12:
Cograts! https://hackaday.com/2024/03/07/an-esp32-multifactor-totp-generator/

AllanOricil, 2024-03-13:
@rzeldent I think the solution would be to use this TFT_eTouch and TFT_eSPI. It seems these libraries allow both Touch and LCD to work on the same SPI bus. 

With your library, I tried setting both touch and lcd to SPI2_HOST, and let the sd card alone in SPI3_HOST, but it did not work. I also did the opposite, and put both in SPI3_HOST, and left the SD card on SPI2_HOST, and again no success. In both cases the LCD works, and the touch is disabled. As an evidence that both LCD and Touch are on the same SPI bus, see the picture shown below

[image]

Still, I don't know why the SD Card isn't working anywhere after calling `smartdisplay_init()`. This is how I'm configureing the SD card to run exclusively on SPI3_HOST

````c
SPIClass custom_spi = SPIClass(VSPI); // using SPI3_HOST exclusively for the SD card

void init_sd_card_reader(){
  custom_spi.begin(TF_SPI_SCLK, TF_SPI_MISO, TF_SPI_MOSI, TF_CS); // 18, 19, 23, 5 respectively

  if(!SD.begin(TF_CS, custom_spi, 80000000)){
    Serial.println("sd card mount failed");
    return;
  }
  uint8_t cardType = SD.cardType();
  if (cardType == CARD_NONE) {
    Serial.println("No SD card attached");
    return;
  }

  Serial.print("SD Card Type: ");
  if (cardType == CARD_MMC) {
      Serial.println("MMC");
  } else if (cardType == CARD_SD) {
      Serial.println("SDSC");
  } else if (cardType == CARD_SDHC) {
      Serial.println("SDHC");
  } else {
      Serial.println("UNKNOWN");
  }

  Serial.printf("SD Card Size: %lluMB\n", SD.cardSize() / (1024 * 1024));
  Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
  Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
}
````

Inside the setup, before calling `smartdisplay_init()`, I can read/append/write to files without a problem. I'm also sure to close the file with `file.close()` after reading it, so that I can later reopen it without a problem. But whenever I try doing it after calling `smartdisplay_init()`, the same problem happens. Even when using a exclusive SPI bus for the SD card.

This is my board config for the experiment described above

````json
{
  "build": {
    "arduino": {
      "ldscript": "esp32_out.ld"
    },
    "core": "esp32",
    "extra_flags": [
      "'-D ARDUINO_ESP32_DEV'",
      "'-D ESP32_2432S028Rv3'",
      "'-D LCD_WIDTH=240'",
      "'-D LCD_HEIGHT=320'",
      "'-D LVGL_BUFFER_PIXELS=(LCD_WIDTH*LCD_HEIGHT/4)'",
      "'-D LVGL_BUFFER_MALLOC_FLAGS=(MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT)'",
      "'-D GPIO_BCKL=21'",
      "'-D LCD_ST7789_SPI'",
      "'-D ST7789_SPI_HOST=SPI2_HOST'",
      "'-D ST7789_SPI_DMA_CHANNEL=SPI_DMA_CH_AUTO'",
      "'-D ST7789_SPI_BUS_MOSI_IO_NUM=13'",
      "'-D ST7789_SPI_BUS_MISO_IO_NUM=GPIO_NUM_NC'",
      "'-D ST7789_SPI_BUS_SCLK_IO_NUM=14'",
      "'-D ST7789_SPI_BUS_QUADWP_IO_NUM=GPIO_NUM_NC'",
      "'-D ST7789_SPI_BUS_QUADHD_IO_NUM=GPIO_NUM_NC'",
      "'-D ST7789_SPI_BUS_MAX_TRANSFER_SZ=0'",
      "'-D ST7789_SPI_BUS_FLAGS=0'",
[... 70 lines cut ...]
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "debug": {
    "openocd_board": "esp-wroom-32.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "esp32-2432S028Rv3",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.aliexpress.com/item/1005004502250619.html",
  "vendor": "Sunton"
}
````

AllanOricil, 2024-03-13:
@rzeldent finally managed to make the SD card work after `smartdisplay_init()`. However the touch is no longer working. I think it is because the touch and lcd are sharing the same SPI bus, but your library wasn't made for that. 

https://github.com/AllanOricil/esp32-mfa-totp-generator/blob/36ef7bac4e55dc0919e2a6bd2e6332146d0eea06/src/main.cpp#L43-L75

Now, I can receive messages from my server using mqtt, and the folder is created
[image]

rzeldent, 2024-03-14:
Hi Allan,

That's a pity touch is not working anymore. Now no time to look into that. However, maybe defining an esp32-2432S028N (No touch) might be something to prevent interference with touch,,,

AllanOricil, 2024-03-14:
> esp32-2432S028N

@rzeldent no problem. But I need the touch to work. I want to use all 3 peripherals. Would your suggestion enable it? Could you explain it in more details? Thank you

AllanOricil, 2024-03-14:
@rzeldent I will put my project to rest for now. Once you you have time to fix this, I will go back to it. I don't have enough knowledge of microcontrolers like you, so it would take me ages to fix it by myself. Thank you anyway. Hope you find time soon!

plueschpruem, 2024-10-04:
> @rzeldent finally managed to make the SD card work after `smartdisplay_init()`. However the touch is no longer working. I think it is because the touch and lcd are sharing the same SPI bus, but your library wasn't made for that.

Well, it works ONCE.
I can init smartdisplay and read a file in loop() – but only once and then it gives errors again.

xperiments, 2024-11-10:
I am using it with the 5 inch
Not without problems

Mostly related to

https://github.com/espressif/arduino-esp32/pull/10122

So I have my own patched version of SD lib

Also worth to mention that SD.begin fails at least first time, so

```
        while (attempt <= 5)
        {
            if (SD.begin(10, SPI, 2000000U, "/sd", maxFiles))
            { ....
```
Cheers!!