rzeldent/esp32-smartdisplay #144: ESP32-3248S035C Display not rotating

ae9000, 2024-03-15:
Hello,
I am trying to get the library working on the ESP32-3248S035C board that uses the ST7796 display driver and has a resolution of 320x480. In portrait mode, everything works as expected:

[image]

 but when rotated using the `lv_disp_set_rotation(disp, LV_DISP_ROT_90);` command it seems like internally the rotation happens, but the pixels don't get drawn to the display in the right order:

[image]

Here's what the main code looks like:
```
#include <Arduino.h>
#include <esp32_smartdisplay.h>
#include <ui/ui.h>

void setup() {
  // put your setup code here, to run once:
  
  smartdisplay_init(); //setup display and touch
  __attribute__((unused)) auto disp = lv_disp_get_default();
   lv_disp_set_rotation(disp, LV_DISP_ROT_90);
   smartdisplay_led_set_rgb(500*0x01, 300*0x02, 0*0x04);
    ui_init();//don't forget
}

void loop() {
  // put your main code here, to run repeatedly:
  lv_timer_handler();
} 

```
I don't know if I'm doing something wrong, or if there is a bug with the specific setup I'm using.
Thank you for the help.

rzeldent, 2024-03-16:
Hi ae9000,

Could reproduce this! Still unclear why. Will look at this...

rzeldent, 2024-03-16:
Something is not right with the hardware rotation. Have to dig deeper.
For now, you can use software rotation:

in the file esp32_smartdisplay\src\lvgl_panel_st7796_spi.c, change the software rotation to:
```drv->sw_rotate = 1;```
  
So the lvgl_lcd_init function looks like:

```
void lvgl_lcd_init(lv_disp_drv_t *drv)
{
    log_v("drv:0x%08x");

    // Hardware rotation is for now disabled
    drv->sw_rotate = 1;
    drv->rotated = LV_DISP_ROT_NONE;
```

ae9000, 2024-03-16:
That works. Thank you so much.

rzeldent, 2024-03-17:
Should work now in the 2.0.9 relase

closed 2024-03-17 by rzeldent