rzeldent/esp32-smartdisplay #149: Display Flickers top to bottom after touching the screen

gathualex, 2024-03-21:
HeytThe user interface uploaded to the board has a blue background, touching any where within **ESP32-8048S050C** ,the screen flickers from top to bottom and the background color changes to brown colour which is not part of the UI, looks like a color inversion. I 
will appreciate any leads to solving this.
Below is a video showing the error.
https://github.com/rzeldent/esp32-smartdisplay/assets/143402429/7087b1b5-903f-4842-95ea-a0005201f696

gathualex, 2024-03-22:
More context..
These errors are printed continuously on the terminal from the board.
`[ 35376][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35376][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35381][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35406][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35406][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35411][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35436][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35436][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35441][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35467][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35467][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35472][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35497][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35497][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35502][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35533][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35533][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35538][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35563][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35563][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35568][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35593][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35593][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35598][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35623][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35623][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35628][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35653][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35653][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35658][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35683][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35683][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35688][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35713][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35713][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35718][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35743][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35743][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35748][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35774][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35774][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35779][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35804][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35804][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35809][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 35846][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 35846][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 35851][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[... cut ...]
[ 36338][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 36343][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1
[ 36368][V][esp32_smartdisplay.c:105] lvgl_touch_calibration_transform(): disp_drv:0x3fca4cf4, data:0x3fcebbc8
[ 36368][V][esp_touch_gt911.c:213] gt911_read_data(): th:0x3d977b08
[ 36373][V][esp_touch_gt911.c:290] gt911_get_xy(): th:0x3d977b08, x:0x3fcebb2c, y:0x3fcebb2e, strength:0x00000000, point_num:0x3fcebb2b, max_point_num:1`

rzeldent, 2024-03-22:
This is verbose logging. Disable logging in the platform.ini

rzeldent, 2024-03-22:
Unfortunately I do not own a esp32-8048S050C. You might try to slower the PCLK variable...
What version are you using?

gathualex, 2024-03-25:
Hi, i solved it by adjusting the memory available from default to  128U * 1024U on lv_conf.h
`/*Size of the memory available for `lv_mem_alloc()` in bytes (>= 2kB)*/

    #define LV_MEM_SIZE (128U * 1024U)          
/*[bytes]*/`
it's working fine though i have observed it flickers one in a while on its own.

gathualex, 2024-03-25:
> Unfortunately I do not own a esp32-8048S050C. You might try to lower the PCLK variable... "'-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=(12.5*1000000)'",
> 
> What version are you using? Let me know if this helps!

esp32-smartdisplay from main branch
lvgl  version 8.3.6