rzeldent/esp32-smartdisplay #150: ESP32-4848S040CIY1 - ESP_ERR_NOT_SUPPORTED esp_lcd_panel_swap_xy(panel_handle, DISPLAY_SWAP_XY)

SorX14, 2024-03-25:
I have this display - https://www.aliexpress.com/item/1005006335587633.html - and tried setting up the demo project but I get the following repeating:

```
[  5149][I][main.cpp:37] setup(): Board: esp32-4848S040CIY1
[  5149][I][main.cpp:38] setup(): CPU: ESP32-S3 rev0, CPU Freq: 240 Mhz, 2 core(s)
[  5150][I][main.cpp:39] setup(): Free heap: 315188 bytes
[  5156][I][main.cpp:40] setup(): Free PSRAM: 8386279 bytes
[  5161][I][main.cpp:41] setup(): SDK version: v4.4.6-dirty
ESP_ERROR_CHECK failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x4037b6d8
file: ".pio/libdeps/esp32-4848S040CIY1/esp32_smartdisplay/src/lvgl_panel_st7701_par.c" line 104
func: lvgl_lcd_init
expression: esp_lcd_panel_swap_xy(panel_handle, DISPLAY_SWAP_XY)

abort() was called at PC 0x4037b6db on core 1

Backtrace: 0x40377eb2:0x3fceba10 0x4037b6e5:0x3fceba30 0x40381605:0x3fceba50 0x4037b6db:0x3fcebad0 0x42029c80:0x3fcebaf0 0x42029ba2:0x3fcebbf0 0x420028e3:0x3fcebc10 0x4202d942:0x3fcebc50

  #0  0x40377eb2:0x3fceba10 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037b6e5:0x3fceba30 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40381605:0x3fceba50 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x4037b6db:0x3fcebad0 in _esp_error_check_failed at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_err.c:43
  #4  0x42029c80:0x3fcebaf0 in lvgl_lcd_init at .pio/libdeps/esp32-4848S040CIY1/esp32_smartdisplay/src/lvgl_panel_st7701_par.c:96 (discriminator 1)
  #5  0x42029ba2:0x3fcebbf0 in smartdisplay_init at .pio/libdeps/esp32-4848S040CIY1/esp32_smartdisplay/src/esp32_smartdisplay.c:219
  #6  0x420028e3:0x3fcebc10 in setup() at src/main.cpp:43
  #7  0x4202d942:0x3fcebc50 in loopTask(void*) at C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
  ```
  
  Commenting `esp32_smartdisplay/src/lvgl_panel_st7701_par.c#L104` then yields:
  
  ```
  [   143][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  5149][I][main.cpp:37] setup(): Board: esp32-4848S040CIY1
[  5149][I][main.cpp:38] setup(): CPU: ESP32-S3 rev0, CPU Freq: 240 Mhz, 2 core(s)
[  5150][I][main.cpp:39] setup(): Free heap: 315188 bytes
[  5156][I][main.cpp:40] setup(): Free PSRAM: 8386279 bytes
[  5161][I][main.cpp:41] setup(): SDK version: v4.4.6-dirty
ESP_ERROR_CHECK failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x4037b6d8
file: ".pio/libdeps/esp32-4848S040CIY1/esp32_smartdisplay/src/lvgl_panel_st7701_par.c" line 107
func: lvgl_lcd_init
expression: esp_lcd_panel_mirror(panel_handle, DISPLAY_MIRROR_X, DISPLAY_MIRROR_Y)

abort() was called at PC 0x4037b6db on core 1

Backtrace: 0x40377eb2:0x3fceba10 0x4037b6e5:0x3fceba30 0x40381605:0x3fceba50 0x4037b6db:0x3fcebad0 0x42029c78:0x3fcebaf0 0x42029b9a:0x3fcebbf0 0x420028db:0x3fcebc10 0x4202d922:0x3fcebc50

  #0  0x40377eb2:0x3fceba10 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037b6e5:0x3fceba30 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40381605:0x3fceba50 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x4037b6db:0x3fcebad0 in _esp_error_check_failed at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_err.c:43
  #4  0x42029c78:0x3fcebaf0 in lvgl_lcd_init at .pio/libdeps/esp32-4848S040CIY1/esp32_smartdisplay/src/lvgl_panel_st7701_par.c:96 (discriminator 1)
  #5  0x42029b9a:0x3fcebbf0 in smartdisplay_init at .pio/libdeps/esp32-4848S040CIY1/esp32_smartdisplay/src/esp32_smartdisplay.c:219
  #6  0x420028db:0x3fcebc10 in setup() at src/main.cpp:43
  #7  0x4202d922:0x3fcebc50 in loopTask(void*) at C:/Users/xxx/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
  ```
  
Commenting `esp32_smartdisplay/src/lvgl_panel_st7701_par.c#L107` then allows the demo to run.
    
Seems like ESP32-4848S040CIY1 does not support these functions.

SorX14, 2024-03-25:
Related? https://github.com/rzeldent/esp32-smartdisplay/issues/147

rzeldent, 2024-03-25:
Hi Steve,

Very probable. Pls check the latest version/develop as this should have been fixed!

Thanks for noticing!

SorX14, 2024-03-25:
Might be missing something, but this doesn't seem to have changed in `develop` - https://github.com/rzeldent/esp32-smartdisplay/blob/develop/src/lvgl_panel_st7701_par.c

winitboonkliang, 2024-03-27:
same issue , I try develop version it not work
please share ESP32-4848S040CIY1 project

danteblank, 2024-03-29:
Right now the only way to make it work is to comment out these lines in esp32_smartdisplay/src/lvgl_panel_st7701_par.c
[image]

emiliozelione, 2024-05-29:
Hi!
I have the same error but in ESP-8048S050. I will try what you did with DISPLAY_SWAP_XY.

closed 2024-12-27 by rzeldent