rzeldent/esp32-smartdisplay #151: Remove demo image

SorX14, 2024-03-25:
Love the library, but can the Lenna image be removed from the demo? She has requested that her image is not used and seems like a small favour to fullfill.

https://en.wikipedia.org/wiki/Lenna

Great yt doc on the image itself: https://www.youtube.com/watch?v=yCdwm2vo09I

rzeldent, 2024-03-25:
Wil have to find a new image; any suggestions?

SorX14, 2024-03-25:
Depends on the intention, if its meant to highlight a screens capability of showing an image, maybe a test card? https://en.wikipedia.org/wiki/Test_card

https://en.wikipedia.org/wiki/Test_card#/media/File:SMPTE_Color_Bars.svg

Or maybe something AI if you're looking for a face? https://generated.photos/faces

closed 2024-12-27 by rzeldent