rzeldent/esp32-smartdisplay #158: ESP32-8048S043C  sdcard configuration for plattformio

joekam54, 2024-04-15:
Hi, first thank you for the great work you have done! here.

I would like to play mjp and jpeg file from the sdcard.  How shouldI set up the sdcard config for platformio? 
Your helpp is very appriciated.

rzeldent, 2024-04-16:
Hi Joe,

There is a duplicate issue about this and so far have not found a solution as this hardware related...

joekam54, 2024-04-17:
Hi Rene,
Thank you! This s a very good input, for now I do not need touch so maybe somehow getting SDcard working will help, because I would like to play out pictures and Video on this Display from the SD card.

plueschpruem, 2024-10-04:
I second this!
Not being able to use the SD is kinda sad since this is otherwise a very nice library!

rzeldent, 2024-10-06:
Take a look at (or continue at)https://github.com/rzeldent/esp32-smartdisplay/issues/143

closed 2024-10-06 by rzeldent