rzeldent/esp32-smartdisplay #159: 8048S070C - external I2C bus unusable due to conflict with GT911_TOUCH_CONFIG_INT_GPIO_NUM

mhaberler, 2024-04-16:
using a  8048S070C / GT911 cap touch board with external I2C sensors connected

it seems the GT911 touch IRQ line (GT911_TOUCH_CONFIG_INT_GPIO_NUM, 18) conflicts with the second I2C Bus SCL line - rendering the second I2C unit  inoperable

I tried in a different project using LovyanGFX and polled touch, and had Wire1 working fine there

not sure how to proceed - not sufficiently familar with ESP-IDF + BSP

thanks in advance,
Michael

rzeldent, 2024-04-17:
If you know where to find it! It is...

However, am wondering, as the IRQ is not used (LVGL is polling) maybe I can leave it out of the configuration...
With your fix, the touch still works right?!

mhaberler, 2024-04-17:
yes touch works perfectly fine