rzeldent/esp32-smartdisplay #160: Support for JC4827W543C (successor of Sunton 4827S043C)

ohlolol, 2024-04-24:
Hi Folks,

i just got my new Display i think its a new revision of the 4827S043C.

details of the board in my other post:
https://github.com/HASwitchPlate/openHASP/issues/713

Arduino_DataBus bus = new Arduino_ESP32QSPI( 45 / cs /, 47 / sck /, 21 / d0 /, 48 / d1 /, 40 / d2 /, 39 / d3 */);

would be nice

ange007, 2024-06-23:
I am also waiting for support. This screens are essentially like a CYD and i want to use one set of libraries.
Unfortunately I was unable to understand the configuration parameters.

rzeldent, 2024-06-25:
Hi,

I made a quick attempt, as I do not own this board, so this might also be a goog starting point for you..

The changes are in the branch nv3041a of https://github.com/rzeldent/esp32-smartdisplay and https://github.com/rzeldent/platformio-espressif32-sunton.

Maybe this works?

ange007, 2024-06-26:
Hi @rzeldent , thanks.
I try build this version.

1. Wrong declared variables: NV3**40**1A_ => NV3**04**1A_ 
2. Class for other lvgl version?
[image]

rzeldent, 2024-06-27:
Fixed these issues. Am/Was having some issues with PlatformIO handing and unable to compile locally....

01GOD, 2024-07-03:
Been trying to get any of the demos that board came with to function. Sigh...seemingly did everything in the PDF it came with, but the Arduino IDE still has sadly shown many errors when trying to compile anything. And yes, did put the lv_config.h into the main dir of the libraries dir. Is it necessary to copy paste that lv_config file from a specific demo folder to main Arduino lib folder when compiling a specific demo's .ino? 

Thanks in advance for answering soon and posting any relevant links about how to use that board.

prashantchawla, 2024-07-21:
@01GOD The demos for the board are only compatible with lvgl version 8 (and not 9). Can you try to install version 8 from arduino libraries? By default it installs version 9.

01GOD, 2024-07-27:
> @01GOD The demos for the board are only compatible with lvgl version 8 (and not 9). Can you try to install version 8 from arduino libraries? By default it installs version 9.

Saw now. Been visiting china and moved to another city and was sadly attacked by crowds of hateful chinese yelling anti-non-chinese profanity and so on, and My body was also violently attacked by four of those chinese men. 

I have been using LVGL 8.3.3 for a while. I can try flashing the newest edition of [esp32-smartdisplay] today. 

Had found out part of the issue was that the shop I bought that board from in china supplied the wrong zip file with it. So after that got the correct zip file and tried the 3 LVGL demos that do thankfully function.

rzeldent, 2024-09-08:
Terrible to hear that. Hope you're well and take care!

And nice to hear it works now!

ange007, 2024-09-09:
@rzeldent 
I do not know what the conversation but this fee is still not working _(and cannot for the reason I pointed above - the wrong description of the board)_.  I added PR for compilation (although the board still does not start as it should).