rzeldent/esp32-smartdisplay #161: esp32-4848S040CIY3 flickering screen will performing update

d4rkdpg, 2024-04-26:
Im using esp32-4848S040CIY3, whilst performing an update the screen is flickering.  I using LVGL and calling a function to do OTA.   I also get a slight flicker when connecting to wifi, or it seems so, not sure whats causing this.

Thanks

rzeldent, 2024-04-26:
Think interrupts are disabled during wifi, or running on the same core as wifi?

d4rkdpg, 2024-04-26:
im not pinning connect to wifitask to a core, the update is running on core0 as a task that gets deleted when update found.

d4rkdpg, 2024-04-29:
tired a few things nothing seems to work, any pointers.

rzeldent, 2024-04-29:
Tweaking the PCLK might also help reducing the flicker..

rzeldent, 2024-05-07:
Any progress?

d4rkdpg, 2024-05-10:
didnt get anywhere still flickering....

rzeldent, 2024-06-14:
The screen is refreshed using I2S. I suspect wifi influences the timing and causes flickering...

d4rkdpg, 2024-06-17:
ok thanks, anything i can do i tried a few things still the same.

xperiments, 2024-07-18:
I have found the same problem

Here is explained and solved but needs changes in ESP IDF

https://forum.lvgl.io/t/display-glitching-while-connecting-to-wifi-esp32s3-8048s043/12075/33

@rzeldent Do you think we can integrate this? checked for bounce_buffer_size_bytes in current version but I imagine the used IDF version is 4.4

closed 2026-10-09 by github-actions[bot]